morrolinux/mpradio-py #21: Should I use mpradio-py or mpradio?

Benji-Collins, 2019-04-18:
If I only want to be able to send audio over bluetooth from my phone, then transmit that audio over FM radio, should I use this newer python version or stick with the legacy version? Thanks.

morrolinux, 2019-04-18:
New version is still under heavy development. However, the feature you want to use is ready and provably already working more reliably than in `mpradio` so I would say go for `mpradio-py`. Plus, I need testers so if you're willing to report bugs and help gathering info about them, even better!
You should really install `mpradio-py` on the latest raspbian version, minimal image, after your normal setup (`raspi-config` etc..)

Benji-Collins, 2019-04-18:
Thanks for your reply. I tried to use mpradio-py and mpradio, however I had/am having the same issue with both on my RPi Zero W.

I am able to find the mpradio Bluetooth device, connect and play music to it, but I am unable to receive any signal on my chosen FM frequency, even with a long wire attached to GPIO 4. Which service is responsible for the FM radio output? I ran `systemctl status mpradio` but that service had stopped stating that a Bluetooth device could not be found. Thanks for any advice.

PS I’m currently using the non python version but can switch back if needed.

Benji-Collins, 2019-04-18:
I just downloaded PiFmAdv by itself and compiled/ran that. I played one of the included test files and it worked perfectly. So this seems to be an issue with transferring the audio from the Bluetooth connection to the pi_fm_adv program.

morrolinux, 2019-04-18:
> PS I’m currently using the non python version but can switch back if needed.
Please do. I'm not supporting the old version anymore. I need you to test on the python version.

> I just downloaded PiFmAdv by itself and compiled/ran that. I played one of the included test files and it worked perfectly. So this seems to be an issue with transferring the audio from the Bluetooth connection to the pi_fm_adv program. I'm going to try the pre-made image for the non-python version to see if that works.

You should really try `systemctl status mpradio` on a fresh raspbian image (where the old mpradio was never installed)

Benji-Collins, 2019-04-18:
I used the latest pre-setup image from the old version and it works perfectly. I’ll try a reinstall with the python version at some point, probably next week when I have better internet. I must have been doing something wrong during setup.

Iocere, 2019-04-19:
The installer actually does not place the python file linked in the systemd

morrolinux, 2019-04-19:
@Iocere Oh yeah that's right I forgot to symlink it

Benji-Collins, 2019-04-22:
Initial question was answered

closed 2019-04-22 by Benji-Collins