morrolinux/mpradio-py #28: Bluetooth connection issue

Iocere, 2019-04-21:
Raspberry 3b+ 
Fresh raspbian stretch lite image

After pairing a bluetooth device the audio (local playback/mp3) get stuck (looping/skipping the current playback) and does not switch over to the BT stream.
The device got paired as far as i can see?

I need to add that i changed the simple-agent service to restart since it fail to run after boot.
That should be in the syslog aswell.

syslog
[https://gist.github.com/Iocere/7b48696a85fad8c2e4522a7a3f2be4b5](https://gist.github.com/Iocere/7b48696a85fad8c2e4522a7a3f2be4b5)

bluetooth_dev
`[Sun Apr 21 22:55:42 BST 2019]: Other action  add
`

daemon.log
[https://gist.github.com/Iocere/95f3027169ea0a041729884b1697d03e](https://gist.github.com/Iocere/95f3027169ea0a041729884b1697d03e)

Iocere, 2019-04-21:
Meh cant reproduce it again. Working now. Changed nothing.

morrolinux, 2019-04-22:
> I need to add that i changed the simple-agent service to restart since it fail to run after boot.

Umh I see... `simple-agent` was trying to start before `bluetooth.target` so I'll need to specify that in the unit file to avoid this scenario. Anyway can you post your modified `simple-agent.service` as well?

> `[Sun Apr 21 22:55:42 BST 2019]: Other action add `

This happens when the `udev` rule gets triggered by the bluetooth connection, but the client's mac address is not being populated in the env for some reason. I could try running a custom python script to catch the client mac address in this scenario. 

About the music getting stuck: now it should just get silent. It's a buffer problem in `pi_fm_rds/pi_fm_adv` in which the data you write gets looped if no new data is coming. a simple workaround is to stream ~1 second of silence before the stop

morrolinux, 2019-04-22:
@Iocere I've implemented the python workaround. You should try to `git pull` and `cd install && sudo bash install.sh` and let me know

Iocere, 2019-04-22:
Tested latest source and the audio still got stuck when switching over to BT.

Simple agent service was stopped.

_Apr 22 22:46:14 raspberrypi bash[586]: Traceback (most recent call last):
Apr 22 22:46:14 raspberrypi bash[586]:   File "/usr/local/bin/simple-agent", line 140, in <module>
Apr 22 22:46:14 raspberrypi bash[586]:     agent.startPairing()
Apr 22 22:46:14 raspberrypi bash[586]:   File "/usr/local/bin/simple-agent", line 118, in startPairing
Apr 22 22:46:14 raspberrypi bash[586]:     adapter_path = findAdapter().object_path
Apr 22 22:46:14 raspberrypi bash[586]:   File "/usr/local/bin/simple-agent", line 48, in findAdapter
Apr 22 22:46:14 raspberrypi bash[586]:     raise Exception("Bluetooth adapter not found")
Apr 22 22:46:14 raspberrypi bash[586]: Exception: Bluetooth adapter not found
Apr 22 22:46:14 raspberrypi systemd[1]: Starting Hostname Service...
Apr 22 22:46:14 raspberrypi systemd[1]: simple-agent.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 22:46:14 raspberrypi systemd[1]: simple-agent.service: Unit entered failed state.
Apr 22 22:46:14 raspberrypi systemd[1]: simple-agent.service: Failed with result 'exit-code'_

A restart of the service later on fixed it tho like last time hence i added 
`Restart=always
RestartSec=3`
to the service even tho thats an ugly approach..guess a late start like` After=multi-user.target` could fix that?

bluetooth_dev now logged just `[Mon Apr 22 22:51:13 BST 2019]: Stop Bluetooth played connection  A0:57:E3:3E:2D:F5
` wich is fine :D

syslog still `Apr 22 22:49:43 raspberrypi systemd-udevd[977]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
` but idk if that connected to the radio.

Did a test run for a few hours today and got 2 crashes. Will try to grab a log if/when it happen again.

Thing I noticed that on a power loss the resume/playlist stuff might get corrupted and will crash the script upon next launch. Maybe implement some integrity checks and restore a new batch of the files.

morrolinux, 2019-04-23:
> to the service even tho thats an ugly approach..guess a late start like` After=multi-user.target` could fix that?

Actually `After=bluetooth.target` would work best I think

> syslog still `Apr 22 22:49:43 raspberrypi systemd-udevd[977]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1. ` but idk if that connected to the radio.

Umh I don't know what that is.

> Did a test run for a few hours today and got 2 crashes. Will try to grab a log if/when it happen again.

Best way of catching logs ATM I think would be to: 
1. Disable mpradio service and reboot
2. SSH into the pi and run mpradio manually
3. wait for the errors to spawn on the shell

> Thing I noticed that on a power loss the resume/playlist stuff might get corrupted and will crash the script upon next launch. Maybe implement some integrity checks and restore a new batch of the files.

Right!

> Overall there seem to be some hiccups since im experiencing the sound freezes/loops after a system reboot without actively connecting via BT aswell.

Not really sure what's wrong there.. more infos needed for sure

Iocere, 2019-04-24:
OK the BT connection didnt drop but the FM went down after a bit. No error logs in the mpradio or the simple-agent console. Got a way to debug the FM ?

morrolinux, 2019-04-24:
@Iocere If you edit `fm_output.py` and remove `stderr=subprocess.PIPE` you'll be able to read pi_fm_adv error log on shell as it happens. It probably won't log anything useful though, as I suspect this might be an undervoltage issue of the Pi3. You should really check for `undervoltage detected` in your kernel logs while fm streaming.. best way to notice it would be to simply connect a monitor via HDMI (`undervoltage detected!` will appear on screen, along with a thunder logo at the top right of the screen)
but I guess it will also appear if you follow kernel logs with `journalctl -kf`

morrolinux, 2019-04-26:
So I guess I'm gonna close this for now, as the original issue is solved anyway. feel free to open another issue for addressing more specifically other issues as well

closed 2019-04-26 by morrolinux