morrolinux/mpradio-py #29: Not working on latest version

Benji-Collins, 2019-04-22:
I have no idea what's going wrong here. What I have done:

1. Installed the premade ISO from the old mpradio - this worked fine.
2. Ran apt upgrade on that install - it broke so I reinstalled.
3. Clean install with latest Raspbian and mpradio-py - install.sh runs fine, can connect via bluetooth but no FM output.

Here's the results of `service mpradio status` to start us off:
```
root@raspberrypi:~# service mpradio status
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-04-22 19:50:01 ACST; 5min ago
  Process: 446 ExecStart=/bin/bash -l -c /home/pi/mpradio.py (code=exited, status=127)
 Main PID: 446 (code=exited, status=127)
Apr 22 19:49:56 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Apr 22 19:50:01 raspberrypi bash[446]: /etc/profile: line 4: id: command not found
Apr 22 19:50:01 raspberrypi bash[446]: /etc/profile: line 4: [: : integer expression expected
Apr 22 19:50:01 raspberrypi bash[446]: /bin/bash: /home/pi/mpradio.py: No such file or directory
Apr 22 19:50:01 raspberrypi systemd[1]: mpradio.service: Main process exited, code=exited, status=127/n/a
Apr 22 19:50:01 raspberrypi systemd[1]: mpradio.service: Unit entered failed state.
Apr 22 19:50:01 raspberrypi systemd[1]: mpradio.service: Failed with result 'exit-code'.
```
Then if I run `service mpradio restart`:
```
root@raspberrypi:~# service mpradio status
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-22 19:55:53 ACST; 2s ago
 Main PID: 569 ((bash))
   CGroup: /system.slice/mpradio.service
           └─569 (bash)
Apr 22 19:55:53 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
```
Even with mpradio.service running, the radio does not work. I decided to run `sudo ln -sF /home/pi/mpradio-py/src/mpradio.py /home/pi/mpradio.py`. This fixed the symlink. Now when I view the status of mpradio I see this:
```
root@raspberrypi:~# service mpradio status
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-22 20:14:24 ACST; 9min ago
 Main PID: 452 (python3)
   CGroup: /system.slice/mpradio.service
           ├─452 python3 /home/pi/mpradio.py
           ├─528 sox -t raw -G -b 16 -e signed -c 2 -r 44100 - -t wav - treble -6
           ├─532 sudo /usr/local/bin/pi_fm_adv --freq 95.1 --ctl /tmp/rds_ctl --audio -
           └─540 /usr/local/bin/pi_fm_adv --freq 95.1 --ctl /tmp/rds_ctl --audio -
Apr 22 20:14:24 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Apr 22 20:14:29 raspberrypi bash[452]: /etc/profile: line 4: id: command not found
Apr 22 20:14:29 raspberrypi bash[452]: /etc/profile: line 4: [: : integer expression expected
Apr 22 20:14:35 raspberrypi sudo[532]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/local/bin/pi_fm_adv --freq 95.1 --ctl /tmp/rds_ctl --audio -
Apr 22 20:14:35 raspberrypi sudo[532]: pam_unix(sudo:session): session opened for user root by (uid=0)
```
However, upon tuning to 95.1 FM, I can only hear silence. While this is an improvement on the static from before I fixed the symlink, this is not good. I am now lead to believe that while the bluetooth is working, as is pi_fm_adv, there is no connection between the two.

I'm happy to provide files and run diagnostic commands. Thanks.

morrolinux, 2019-04-22:
> Apr 22 19:50:01 raspberrypi bash[446]: /bin/bash: /home/pi/mpradio.py: No such file or directory

You should really check if you have a symbolic link in your home which points to /home/mpradio/src/mpradio.py 
Maybe in your home the folder is called `mpradio-py` ? because I haven't specified that you need to `git clone https://github.com/morrolinux/mpradio-py.git mpradio` from your home

Benji-Collins, 2019-04-22:
Yes I cloned mpradio. Further down my comment I mentioned that the symlink did not work correctly. I then fixed it.

morrolinux, 2019-04-22:
Alright I see you got this thing fixed. I wasn't at that part yet.
out of curiosity, does mp3 playback from storage work? for instance if you put an mp3 song in `/pirateradio/` does it work?

can you `tail -f  /var/log/bluetooth_dev` while connecting your phone?

Benji-Collins, 2019-04-22:
Added an mp3 and after restarting I could hear fine. I ran `tail -f /var/log/bluetooth_dev` while connecting from my iPhone:
```
root@raspberrypi:~# tail -f /var/log/bluetooth_dev
[Mon Apr 22 20:35:02 ACST 2019]: Stop Bluetooth played connection  xx.xx.xx.xx.xx.xx
tail: /var/log/bluetooth_dev: file truncated
[Mon Apr 22 22:32:09 ACST 2019]: Start Bluetooth played connection  xx.xx.xx.xx.xx.xx
tail: /var/log/bluetooth_dev: file truncated
[Mon Apr 22 22:32:09 ACST 2019]: Other action  add
```
I tried to play music via bluetooth immediately and it starting working fine, just as it did with my previous installation. I thought that maybe a local mp3 file is needed for bluetooth playback but after removal, bluetooth still works fine. I'm not overly sure why this is.

Benji-Collins, 2019-04-22:
Another issue I have noticed however is that my phone does not auto connect to my Pi and requires me to select it ever time. Would this be an issue with my phone or some sort of 'saved device' list in mpradio?

morrolinux, 2019-04-22:
My phone connects automatically to the Pi when I enable Bluetooth, so I'm not sure..

Benji-Collins, 2019-04-22:
It's probably my phone. I'll see if I can get it to work.

morrolinux, 2019-04-25:
So have you got any feedback on overall usage and stability? bluetooth playback works fine?

Benji-Collins, 2019-04-25:
Everything other than auto-connecting (no big deal for me) seems to be working fine now thanks. Running smoothly on my Pi Zero W and only stutters during other CPU intensive tasks like updates.

closed 2019-04-25 by Benji-Collins