morrolinux/mpradio-py #34: Static between songs

rfancella, 2019-04-26:
Hi,
I ran across this project a couple days ago.  I love it!  Very feature rich and easy to setup compared to any other setup out there.
I installed the most recent Raspbian minimal on a Raspberry PI 0 W and updated it.  I installed using GIT and it fired right up and worked as expected.  Great job there!
One thing I did notice, when switching tracks either manually or when it goes to the next track, there is a second of static as it stops transmitting.  Is that normal or am I missing something?
I do like the fact there is a skip/power down button.  Would it be possible to implement more buttons and possibly a OLED or LCD display on the PI?  Or maybe even a menu system that shows over the RDS also?  Last thought, how about an "RDS" file named the same as the audio file, that pipes through the rds control to display lyrics or other information about the audio file?
I would interested to hear your thoughts on this?

Thank you,
Ron Fancella

morrolinux, 2019-04-26:
Hi, I'm glad you like the project and had no major issues so far. 

> One thing I did notice, when switching tracks either manually or when it goes to the next track, there is a second of static as it stops transmitting. Is that normal or am I missing something?

That sounds strange... are you sure you're not talking about the old mpradio implementation (https://github.com/morrolinux/mpradio) because that was doing it because of its design. 
This python version here is really different in terms of design: the FM transmitter should never stop (even between songs) and is not really capable of re-starting anyway, so that is really strange to hear. Maybe you could also make a small video showing the issue.

> I do like the fact there is a skip/power down button. Would it be possible to implement more buttons and possibly a OLED or LCD display on the PI? Or maybe even a menu system that shows over the RDS [...]

So... more buttons shouldn't be a problem, but they have to be shaped differently for it to make sense, as I plan to use them while driving and my eyes should be on the road. An OLED would be possible as well, but I wouln't put such an effort in it yet, as the project is not really big or mature as of yet. 
The menu system over RDS is quite a good idea, I already thoguht about it and probably gonna make it happen someday. The only issue with that is the car's RDS display response time... many cars like mine lookup for RDS only every 5 seconds, which is soo slow for it to be really usable, so I don't know...
Same applies for the lyrics... I couln't really test it as the refresh would be too slow

morrolinux, 2019-04-26:
Oh right I forgot, you might be happy to know that I'm also working on the Android companion app. Right now it's possible to view the playlist and pick a song to be played immediately, along with playback controls.
Settings management and OBEX FTP will be supported soon as well

rfancella, 2019-04-26:
Thank you so much for the prompt response!
LOL.  Yea, so I'll update it to the mpradio-py and see how it goes.

closed 2019-04-29 by morrolinux