morrolinux/mpradio-py #39: error when starting ./mpradio.py missing 'psutil'

ikoydg, 2019-05-08:
i have this error when starting ./mpradio.py

fresh installed 2019-04-08-raspbian-stretch-full and updated. 

>   Failed building wheel for av
  Running setup.py clean for av
Failed to build av
Installing collected packages: av, psutil
  Running setup.py install for av ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, toke                                                                                                                                                             nize;__file__='/tmp/pip-build-olob6law/av/setup.py';f=getattr(tokenize, 'open',                                                                                                                                                              open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code,                                                                                                                                                              __file__, 'exec'))" install --record /tmp/pip-qfj505wj-record/install-record.tx                                                                                                                                                             t --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/av
    copying av/__main__.py -> build/lib.linux-armv7l-3.5/av
    copying av/datasets.py -> build/lib.linux-armv7l-3.5/av
    copying av/__init__.py -> build/lib.linux-armv7l-3.5/av
    copying av/deprecation.py -> build/lib.linux-armv7l-3.5/av
    creating build/lib.linux-armv7l-3.5/av/codec
    copying av/codec/__init__.py -> build/lib.linux-armv7l-3.5/av/codec
    creating build/lib.linux-armv7l-3.5/av/subtitles
    copying av/subtitles/__init__.py -> build/lib.linux-armv7l-3.5/av/subtitles
    creating build/lib.linux-armv7l-3.5/av/container
    copying av/container/__init__.py -> build/lib.linux-armv7l-3.5/av/container
    creating build/lib.linux-armv7l-3.5/av/audio
    copying av/audio/__init__.py -> build/lib.linux-armv7l-3.5/av/audio
    creating build/lib.linux-armv7l-3.5/av/video
    copying av/video/__init__.py -> build/lib.linux-armv7l-3.5/av/video
    creating build/lib.linux-armv7l-3.5/av/data
    copying av/data/__init__.py -> build/lib.linux-armv7l-3.5/av/data
    creating build/lib.linux-armv7l-3.5/av/filter
    copying av/filter/__init__.py -> build/lib.linux-armv7l-3.5/av/filter
    running build_ext
    running config
    Could not find libavformat with pkg-config.
    Could not find libavcodec with pkg-config.
    Could not find libavdevice with pkg-config.
    Could not find libavutil with pkg-config.
    Could not find libavfilter with pkg-config.
    Could not find libswscale with pkg-config.
    Could not find libswresample with pkg-config.

> Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-                                                                                                                                                             build-8smjkpmb/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.r                                                                                                                                                             ead().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" ins                                                                                                                                                             tall --record /tmp/pip-y9o2qt2c-record/install-record.txt --single-version-exter                                                                                                                                                             nally-managed --compile" failed with error code 1 in /tmp/pip-build-8smjkpmb/av/
Failed to enable unit: Refusing to operate on linked unit file dbus-org.bluez.se                                                                                                                                                             rvice
Synchronizing state of bluetooth.service with SysV service script with /lib/syst                                                                                                                                                             emd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable bluetooth
fatal: destination path 'PiFmAdv' already exists and is not an empty directory.
rm -f *.o
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 rds.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 waveforms.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2   -c -o pi_fm_adv.o pi_fm_adv.                                                                                                                                                             c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 fm_mpx.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 control_pipe.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s                                                                                                                                                              -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 mailbox.c
gcc -o pi_fm_adv rds.o waveforms.o mailbox.o pi_fm_adv.o fm_mpx.o control_pipe.o                                                                                                                                                              -lm -lsndfile

then it restarts after installation

after running  ./mpradio.pi

> pi@raspberrypi:~ $ cd mpradio/src
pi@raspberrypi:~/mpradio/src $ ./mpradio.py
Traceback (most recent call last):
  File "./mpradio.py", line 12, in <module>
    from storage_player import StoragePlayer
  File "/home/pi/mpradio/src/storage_player.py", line 10, in <module>
    import psutil
ImportError: No module named 'psutil'

tried fresh installating 3 times, not working.
Im using  RPI 3 b+ with 8gb memcard.

morrolinux, 2019-05-09:
Hi,
Not really sure about what happened there but... what happens if you `sudo pip3 install psutil` ?

ikoydg, 2019-05-10:
already done that.  `sudo pip3 install psutil`  and `sudo pip3 install av`

still got this error 

>   Failed building wheel for av
  Running setup.py clean for av
Failed to build av
Installing collected packages: av
  Running setup.py install for av ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, toke                                                                                                                                                             nize;__file__='/tmp/pip-build-x7cxhg1w/av/setup.py';f=getattr(tokenize, 'open',                                                                                                                                                              open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code,                                                                                                                                                              __file__, 'exec'))" install --record /tmp/pip-qd4i2sis-record/install-record.tx                                                                                                                                                             t --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/av
    copying av/__main__.py -> build/lib.linux-armv7l-3.5/av
    copying av/datasets.py -> build/lib.linux-armv7l-3.5/av
    copying av/__init__.py -> build/lib.linux-armv7l-3.5/av
    copying av/deprecation.py -> build/lib.linux-armv7l-3.5/av
    creating build/lib.linux-armv7l-3.5/av/codec
    copying av/codec/__init__.py -> build/lib.linux-armv7l-3.5/av/codec
    creating build/lib.linux-armv7l-3.5/av/subtitles
    copying av/subtitles/__init__.py -> build/lib.linux-armv7l-3.5/av/subtitles
    creating build/lib.linux-armv7l-3.5/av/container
    copying av/container/__init__.py -> build/lib.linux-armv7l-3.5/av/container
    creating build/lib.linux-armv7l-3.5/av/audio
    copying av/audio/__init__.py -> build/lib.linux-armv7l-3.5/av/audio
    creating build/lib.linux-armv7l-3.5/av/video
    copying av/video/__init__.py -> build/lib.linux-armv7l-3.5/av/video
    creating build/lib.linux-armv7l-3.5/av/data
    copying av/data/__init__.py -> build/lib.linux-armv7l-3.5/av/data
    creating build/lib.linux-armv7l-3.5/av/filter
    copying av/filter/__init__.py -> build/lib.linux-armv7l-3.5/av/filter
    running build_ext
    running config
    Could not find libavformat with pkg-config.
    Could not find libavcodec with pkg-config.
    Could not find libavdevice with pkg-config.
    Could not find libavutil with pkg-config.
    Could not find libavfilter with pkg-config.
    Could not find libswscale with pkg-config.
    Could not find libswresample with pkg-config.

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-                                                                                                                                                             build-x7cxhg1w/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.r                                                                                                                                                             ead().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" ins                                                                                                                                                             tall --record /tmp/pip-qd4i2sis-record/install-record.txt --single-version-exter                                                                                                                                                             nally-managed --compile" failed with error code 1 in /tmp/pip-build-x7cxhg1w/av/

morrolinux, 2019-05-10:
seems like an issue with pyav installation... have you got `pkg-config` package installed on your system?

morrolinux, 2019-05-13:
so... any news?

rfancella, 2019-05-14:
I have the same setup with the same issue.
pyav will not install.  I have detailed document of the install from start to finish.
I did get psutil installed with no issues.  But av still will not go. Is there some way I can attach the document here?  It's just a text file.

morrolinux, 2019-05-14:
Thanks for the feedback. However I think you should open a ticket on pyav github page, as this issue seems entirely related with pyav install rather than mpradio install

rfancella, 2019-05-14:
Ok, will do.
I think I'll install via source first.  And take it from there.  I'll post my findings.

morrolinux, 2019-05-14:
Yes please, thanks :)

rfancella, 2019-05-14:
Ok, I managed to get av installed with the help of this link:
[https://docs.mikeboers.com/pyav/develop/installation.html](url)
This suggested:
# General dependencies
sudo apt-get install -y python-dev pkg-config

# Library components
sudo apt-get install -y \
    libavformat-dev libavcodec-dev libavdevice-dev \
    libavutil-dev libswscale-dev libswresample-dev libavfilter-dev

Once the above was complete:
sudo pip3 install av

It took a while, but installed.
Rebooted with a USB drive and some mp3's on it.
Still an issue, but it has nothing to do with this issue.  So, I will open another.
Thank you!

morrolinux, 2019-05-14:
Ok, I'm going to update the installer. if you happen to test it on a fresh newly installed system, please let me know if you encounter any more problems

rfancella, 2019-05-14:
I'm on it.  Fresh install going now.
:D

morrolinux, 2019-05-14:
I just added `python3-dev` and `pkg-config` as I noticed the other packages were not installed on my working system, so I guess they're not really needed

rfancella, 2019-05-14:
Did you fix the symlink as well?
Upon running the install it creates:
lrwxrwxrwx 1 root root   14 May 14 11:15 mpradio.py -> src/mpradio.py
Instead of:
lrwxrwxrwx 1 root root   22 May 14 11:19 mpradio.py -> mpradio/src/mpradio.py

This go-round I did install all the packages I suggested above.  I'll try it with just the ones you suggested.

morrolinux, 2019-05-14:
Ok! let me know then :)
ps: I haven't touched the symlink

rfancella, 2019-05-14:
Ok, it still fails on av.
If you install:
sudo apt-get install -y \
    libavformat-dev libavcodec-dev libavdevice-dev \
    libavutil-dev libswscale-dev libswresample-dev libavfilter-dev
It does complete the install without errors.
But this is where it goes very wrong.  ;)
See attached: 
[MPRadio Install - Take Three.txt](https://github.com/morrolinux/mpradio-py/files/3179033/MPRadio.Install.-.Take.Three.txt)
[image]

morrolinux, 2019-05-14:
So... something must have changed since I last installed the Pi. Otherwise I can't explain why I have none of the packages you mentioned installed, and yet pyav installs fine and works. I guess I'll just update the installer accordingly.
As for the post install going very wrong: can you provide your `/etc/crontab` file?

rfancella, 2019-05-14:
Yes sir.

proc            /proc           proc    defaults          0       0
PARTUUID=9bbb34f0-01  /boot           vfat    defaults          0       2
PARTUUID=9bbb34f0-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
/dev/sda1    /pirateradio    vfat    defaults,rw,uid=pi,gid=pi,nofail 0   0

morrolinux, 2019-05-14:
It looks fine... What's strange about the boot picture you attached is that... `/pirateradio` mount point fails probably because you haven't attached a FAT32 single-partitioned usb drive, and that's ok... The `nofail` option I specified for that mount point should prevent the system to go into recovery filesafe.
The thing is, it also fails mounting `/boot` which is not normal. Are you sure your SD haven't become corrupted?

rfancella, 2019-05-14:
Yes, the /boot partition was corrupt.  This happened every time (three attempts to install fresh).  I removed the dirty bit.  It runs now!  Broadcasting as expected.
NOTE: I did have to remove the /home/pi/mpradio.py and symlink it before it would start.
Great job there!  I will have tons of fun with it.  I plan to mount it in my car.  Copy the contents of the usb stick into /pirateradio and remove the fstab entry so it will run without a usb drive.
Then, create an auto sync (rsync) when it gets close enough to my home wifi so I just update a directory on my server.  Connect a small battey to it, tie a GPIO pin to the ignition and have it do a safe shutdown when the car gets turned off.  Just some ideal stuff for my application.  :)
Again, great job getting this working!

morrolinux, 2019-05-14:
> NOTE: I did have to remove the /home/pi/mpradio.py and symlink it before it would start.

That is really awkward. it should have been a working symlink already. I'll probably change that anyways.

> Then, create an auto sync (rsync) when it gets close enough to my home wifi so I just update a directory on my server. Connect a small battey to it, tie a GPIO pin to the ignition and have it do a safe shutdown when the car gets turned off. Just some ideal stuff for my application. :)

What an awesome collection of great ideas you've got here! Cool, I hope you'll release it as well :)

closed 2019-05-14 by morrolinux