morrolinux/mpradio-py #44: no sound from Bluetooth on Raspberry Pi Zero W

Fischman132, 2019-06-20:
[output.txt](https://github.com/morrolinux/mpradio-py/files/3309743/output.txt)

yaneek1337, 2019-07-01:
Same for me.

morrolinux, 2019-07-01:
Try the following:
1. Switch to no_buffer_to_output brach
2. Run the installation again
It's experimental but and under development, so I can't guarantee it won't break, but it should work
Let me know!

yaneek1337, 2019-07-01:
Where to switch? Thanks for reply :)

morrolinux, 2019-07-01:
Oh sorry. I've updated my last comment

princeminz, 2019-07-04:
no_buffer_to_output branch doesn't work for me on rpi zero w, can't hear anything(no sound).
Also where can I find the usage instructions for this program? (are they the same as mpradio repository?)

morrolinux, 2019-07-04:
@princeminz Try the following:
1. Put an mp3 song in `/pirateradio`
2. Reboot and retry
PS: For some reason bluetooth playback might break until reboot after the first attempt gone wrong

princeminz, 2019-07-05:
Tried 5 times, didn't work, reinstalled the os and then tried still didn't work. Getting this [output](https://github.com/morrolinux/mpradio-py/files/3362660/output.txt) .The audio clip i placed in the folder plays but the bluetooth is not working.

morrolinux, 2019-07-05:
Looks like the control pipe in `/tmp/mpradio_bt` is not being read correctly...
can you post your `/var/bluetooth_dev` log?

princeminz, 2019-07-05:
can't find /var/bluetooth_dev

```bash
pi@raspberrypi:/var $ cd /var
pi@raspberrypi:/var $ ls
backups  cache  lib  local  lock  log  mail  opt  run  spool  swap  tmp
pi@raspberrypi:/var $
```

princeminz, 2019-07-05:
Something changed in the log this time i ran mpradio.py but I still can't hear anything. Here is the log
```
pi@raspberrypi:~ $ ./mpradio.py
mpradio main PID is: 1294
broadcasting on FM 88.0
storage_player playing: /pirateradio/a.mp3
/home/pi/mpradio/src/gpio_remote.py:36: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(5, GPIO.IN, pull_up_down=GPIO.PUD_UP)
Estimating duration from bitrate, this may be inaccurate
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib bluealsa-pcm.c:679:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device or address
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
playing bluetooth: 0C:9D:92:91:89:8D
bluetooth attached
skipping flush...

```

jonrietveld, 2019-08-13:
Same issue here, it seems to be an issue with the alsa audio server and the bluealsa library. I read somewhere alsa is not supported on newer versions of raspbian.

For me the transmitter transmits silence and I can connect to the Bluetooth, but no audio is being transmitted. This is tested on the newest version of raspbian buster lite.

jonrietveld, 2019-08-15:
What is the linux distro you know this code will work with? Do you happen to have an image file? (I have a raspberry pi zero w)

When I run with the latest version of raspbian, raspbian buster lite, I get the following message in syslog:
```
Aug 15 23:05:43 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 23:05:43 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
```
Also when I run journalctl I get:
```
pi@raspberrypi:~ $ sudo journalctl -u mpradio
-- Logs begin at Thu 2019-08-15 23:02:52 BST, end at Thu 2019-08-15 23:28:18 BST. --
Aug 15 23:03:19 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Aug 15 23:03:24 raspberrypi bash[459]: /etc/profile: line 4: id: command not found
Aug 15 23:03:24 raspberrypi bash[459]: /etc/profile: line 4: [: : integer expression expected
Aug 15 23:03:35 raspberrypi sudo[485]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/local/bin/pi_fm_adv --freq 87.9 --ctl /tmp/rds_ctl --audio -
Aug 15 23:03:35 raspberrypi bash[459]: /home/pi/mpradio/src/gpio_remote.py:36: RuntimeWarning: A physical pull up resistor is fitted on this channel!
Aug 15 23:03:35 raspberrypi bash[459]:   GPIO.setup(5, GPIO.IN, pull_up_down=GPIO.PUD_UP)
Aug 15 23:03:35 raspberrypi bash[459]: mpradio main PID is: 459
Aug 15 23:03:35 raspberrypi bash[459]: broadcasting on FM 87.9
Aug 15 23:03:35 raspberrypi bash[459]: Media scanner scanning folder: /pirateradio/
Aug 15 23:03:35 raspberrypi bash[459]: ERR: No songs to play
Aug 15 23:03:35 raspberrypi sudo[485]: pam_unix(sudo:session): session opened for user root by (uid=0)
```
also for the other services:
```
pi@raspberrypi:~ $ sudo journalctl -u mpradio-bt-setup.service
-- Logs begin at Thu 2019-08-15 23:02:52 BST, end at Thu 2019-08-15 23:31:20 BST. --
Aug 15 23:03:07 raspberrypi systemd[1]: Starting Mpradio bluetooth setup...
Aug 15 23:03:19 raspberrypi bash[371]: Serial Port service registered
Aug 15 23:03:19 raspberrypi systemd[1]: Started Mpradio bluetooth setup.
pi@raspberrypi:~ $ sudo journalctl -u need2recompile.service
-- Logs begin at Thu 2019-08-15 23:02:52 BST, end at Thu 2019-08-15 23:31:38 BST. --
Aug 15 23:03:19 raspberrypi systemd[1]: Started Determine wheather the running device has changed and PiFmAdv recompile is needed or not.
Aug 15 23:03:24 raspberrypi systemd[1]: need2recompile.service: Succeeded.
pi@raspberrypi:~ $ sudo journalctl -u simple-agent.service
-- Logs begin at Thu 2019-08-15 23:02:52 BST, end at Thu 2019-08-15 23:31:51 BST. --
Aug 15 23:03:19 raspberrypi systemd[1]: Started Bluealsa audio server.
```
I do not have a flash drive attached so /dev/sda does not exist. Is that why pirateradio is failing? I am not getting any sound out of the the transmission, however my rpi (zero w) is transmitting silence, and the bluetooth is connecting to my phone. Other than the pirateradio failing, I cannot find another relevant services that is failing in syslog.

Here is all failed messages in syslog:

```
pi@raspberrypi:~ $ cat /var/log/syslog | grep 'fail'
Aug 15 22:29:44 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 22:29:44 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 15 22:29:44 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
Aug 15 22:31:37 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 22:31:37 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 15 22:31:37 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
Aug 15 22:32:10 raspberrypi avahi-daemon[263]: socket() failed: Address family not supported by protocol
Aug 15 22:32:10 raspberrypi avahi-daemon[263]: socket() failed: Address family not supported by protocol
Aug 15 22:32:16 raspberrypi bluetoothd[363]: Sap driver initialization failed.
Aug 15 22:33:33 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 22:33:33 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 15 22:33:33 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
Aug 15 22:34:50 raspberrypi systemd-udevd[598]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Aug 15 22:38:13 raspberrypi kernel: [  375.362643] Bluetooth: hci0: sending frame failed (-49)
Aug 15 22:38:35 raspberrypi avahi-daemon[258]: socket() failed: Address family not supported by protocol
Aug 15 22:38:35 raspberrypi avahi-daemon[258]: socket() failed: Address family not supported by protocol
Aug 15 22:38:40 raspberrypi bluetoothd[362]: Sap driver initialization failed.
Aug 15 22:53:13 raspberrypi systemd-udevd[510]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Aug 15 22:53:50 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 22:53:50 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 15 22:53:50 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
Aug 15 22:56:06 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 22:56:06 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
[... cut ...]
Aug 15 23:03:09 raspberrypi avahi-daemon[275]: socket() failed: Address family not supported by protocol
Aug 15 23:03:10 raspberrypi avahi-daemon[275]: socket() failed: Address family not supported by protocol
Aug 15 23:03:16 raspberrypi bluetoothd[362]: Sap driver initialization failed.
Aug 15 23:05:43 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 15 23:05:43 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 15 23:05:43 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
```

morrolinux, 2019-08-16:
@jonrietveld you are looking at the invocation "stack" so that's why you see pi_fm_adv two times (one preceeded by sudo and the other not). I haven't tested Raspbian 10 as of yet. You should try with the previous one, Raspbian 9, which I'm developing on

jonrietveld, 2019-08-16:
I just fresh installed raspbian stretch from here:
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/
ran the install script and blacklisted the modules listed, and still just silence on the radio. Here is what failed in syslog:

```
pi@raspberrypi:~ $ cat /var/log/syslog | grep fail
Apr  8 11:31:15 raspberrypi systemd-udevd[124]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
Apr  8 11:31:17 raspberrypi systemd-udevd[122]: Process '/sbin/crda' failed with exit code 249.
Apr  8 11:31:17 raspberrypi systemd-udevd[122]: Process '/sbin/crda' failed with exit code 249.
Apr  8 11:31:21 raspberrypi bluetoothd[412]: Sap driver initialization failed.
Aug 16 13:39:41 raspberrypi lightdm[378]: session_get_login1_session_id: assertion 'session != NULL' failed
Aug 16 14:14:33 raspberrypi kernel: [ 2151.217359] Bluetooth: hci0 sending frame failed (-49)
Aug 16 14:14:47 raspberrypi avahi-daemon[229]: socket() failed: Address family not supported by protocol
Aug 16 14:14:47 raspberrypi avahi-daemon[229]: socket() failed: Address family not supported by protocol
Aug 16 14:14:52 raspberrypi systemd-udevd[124]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:14:52 raspberrypi bluetoothd[387]: Sap driver initialization failed.
Aug 16 14:14:52 raspberrypi systemd-udevd[126]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:14:57 raspberrypi systemd-udevd[127]: Process '/usr/lib/udev/bluetooth' failed with exit code 1.
Aug 16 14:14:59 raspberrypi bash[461]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:15:05 raspberrypi bash[561]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:15:05 raspberrypi bash[564]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:16:04 raspberrypi kernel: [   82.185705] Bluetooth: hci0 sending frame failed (-49)
Aug 16 14:16:19 raspberrypi avahi-daemon[242]: socket() failed: Address family not supported by protocol
Aug 16 14:16:19 raspberrypi avahi-daemon[242]: socket() failed: Address family not supported by protocol
Aug 16 14:16:23 raspberrypi systemd-udevd[123]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:16:24 raspberrypi systemd-udevd[123]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:16:25 raspberrypi bluetoothd[385]: Sap driver initialization failed.
Aug 16 14:16:29 raspberrypi systemd-udevd[124]: Process '/usr/lib/udev/bluetooth' failed with exit code 1.
Aug 16 14:16:31 raspberrypi bash[430]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:16:36 raspberrypi bash[493]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:16:36 raspberrypi bash[490]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:18:45 raspberrypi systemd-udevd[722]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Aug 16 14:19:34 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 16 14:19:34 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 16 14:19:34 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
Aug 16 14:20:30 raspberrypi systemd-udevd[1679]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Aug 16 14:23:18 raspberrypi avahi-daemon[238]: socket() failed: Address family not supported by protocol
Aug 16 14:23:19 raspberrypi avahi-daemon[238]: socket() failed: Address family not supported by protocol
Aug 16 14:23:24 raspberrypi bluetoothd[376]: Sap driver initialization failed.
Aug 16 14:23:24 raspberrypi systemd-udevd[134]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:23:25 raspberrypi systemd-udevd[124]: Process '/sbin/crda' failed with exit code 249.
Aug 16 14:23:29 raspberrypi systemd-udevd[120]: Process '/usr/lib/udev/bluetooth' failed with exit code 1.
Aug 16 14:23:31 raspberrypi bash[434]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:23:36 raspberrypi bash[494]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:23:36 raspberrypi bash[496]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:26:49 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 16 14:26:49 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 16 14:26:49 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.pi@raspberrypi:~ $ cat /var/log/syslog | grep fail
Apr  8 11:31:15 raspberrypi systemd-udevd[124]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
Apr  8 11:31:17 raspberrypi systemd-udevd[122]: Process '/sbin/crda' failed with exit code 249.
Apr  8 11:31:17 raspberrypi systemd-udevd[122]: Process '/sbin/crda' failed with exit code 249.
Apr  8 11:31:21 raspberrypi bluetoothd[412]: Sap driver initialization failed.
Aug 16 13:39:41 raspberrypi lightdm[378]: session_get_login1_session_id: assertion 'session != NULL' failed
Aug 16 14:14:33 raspberrypi kernel: [ 2151.217359] Bluetooth: hci0 sending frame failed (-49)
Aug 16 14:14:47 raspberrypi avahi-daemon[229]: socket() failed: Address family not supported by protocol
Aug 16 14:14:47 raspberrypi avahi-daemon[229]: socket() failed: Address family not supported by protocol
Aug 16 14:14:52 raspberrypi systemd-udevd[124]: Process '/sbin/crda' failed with exit code 249.
[... cut ...]
Aug 16 14:23:36 raspberrypi bash[496]: mesg: ttyname failed: Inappropriate ioctl for device
Aug 16 14:26:49 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Aug 16 14:26:49 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Aug 16 14:26:49 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.
```

Is there any way I could check if the audio is correctly being received from my phone, or check if pi_fm_adv is transmitting correctly?

morrolinux, 2019-08-16:
I'm not sure what's happening here. I've never seen those services fail and what's `/sbin/crda` btw?
From which Raspberry Pi are you running it? 
The other issue is: I'm on holidays starting tomorrow and I won't be here for about one week, if you wish to debug this (and if you can, please do try! :) you can have a look at the schematic here:
https://github.com/morrolinux/mpradio/blob/master/README.md
It's from the legacy version, but system components are still the same

jonrietveld, 2019-08-16:
Apparently crda has something to do with a wireless regulatory domain whatever that means:
https://linux.die.net/man/8/crda
I just pulled your code from mpradio and installed it on my rpi zero w, and now everything seems to work correctly, so Im not really sure what the issue is. Unfortunatly I am not very good at debugging services, but your old code ran correctly on the same install of raspbian 9 that mpradio-py didn't work with.
Thanks for your help

squadramunter, 2019-09-11:
Same problem here. I run the latest version Buster and I also don't get sound trough the FM Transmitter via Bluetooth.

squadramunter, 2019-09-11:
``
pi@raspberrypi:~ $ cat /var/log/syslog | grep fail
Jul 10 01:21:20 raspberrypi systemd-udevd[145]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 0' failed with exit code 99.
Jul 10 01:21:20 raspberrypi systemd-udevd[144]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Jul 10 01:21:20 raspberrypi systemd-udevd[146]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Jul 10 01:21:20 raspberrypi systemd-udevd[147]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Jul 10 01:21:35 raspberrypi bluetoothd[529]: Sap driver initialization failed.
Sep 11 13:45:10 raspberrypi systemd-udevd[151]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 11 13:45:10 raspberrypi systemd-udevd[150]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 11 13:45:10 raspberrypi systemd-udevd[146]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 11 13:45:15 raspberrypi bluetoothd[604]: Sap driver initialization failed.
Sep 11 13:45:58 raspberrypi systemd-udevd[794]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 11 13:46:18 raspberrypi kernel: [   66.241129] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241191] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241287] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241362] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241448] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241534] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241621] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241710] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241796] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241880] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.241974] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242054] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242146] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242236] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242329] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242415] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242498] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242586] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242676] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242763] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242846] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.242932] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243019] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243111] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243193] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243279] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243366] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243452] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243539] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243626] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243712] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243799] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243886] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.243975] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244059] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244146] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244233] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244319] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244405] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244492] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244579] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:46:18 raspberrypi kernel: [   66.244667] Bluetooth: hci0: Frame reassembly failed (-84)
[... cut ...]
Sep 11 13:51:01 raspberrypi kernel: [  220.540806] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:51:01 raspberrypi kernel: [  220.540848] Bluetooth: hci0: Frame reassembly failed (-84)
Sep 11 13:51:37 raspberrypi systemd[1]: Dependency failed for /pirateradio.
Sep 11 13:51:37 raspberrypi systemd[1]: pirateradio.mount: Job pirateradio.mount/start failed with result 'dependency'.
Sep 11 13:51:37 raspberrypi systemd[1]: dev-sda1.device: Job dev-sda1.device/start failed with result 'timeout'.

``

squadramunter, 2019-09-11:
pi@raspberrypi:~ $ service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-09-11 13:47:24 BST; 11min ago
     Docs: man:bluetoothd(8)
 Main PID: 559 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 2200)
   Memory: 3.9M
   CGroup: /system.slice/bluetooth.service
           └─559 /usr/lib/bluetooth/bluetoothd -C

Sep 11 13:47:24 raspberrypi systemd[1]: Starting Bluetooth service...
Sep 11 13:47:24 raspberrypi bluetoothd[559]: Bluetooth daemon 5.50
Sep 11 13:47:24 raspberrypi systemd[1]: Started Bluetooth service.
Sep 11 13:47:24 raspberrypi bluetoothd[559]: Starting SDP server
Sep 11 13:47:24 raspberrypi bluetoothd[559]: Bluetooth management interface 1.14 initialized
Sep 11 13:47:24 raspberrypi bluetoothd[559]: Sap driver initialization failed.
Sep 11 13:47:24 raspberrypi bluetoothd[559]: sap-server: Operation not permitted (1)
Sep 11 13:47:24 raspberrypi bluetoothd[559]: Failed to set privacy: Rejected (0x0b)
Sep 11 13:47:29 raspberrypi bluetoothd[559]: Endpoint registered: sender=:1.12 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 11 13:47:58 raspberrypi bluetoothd[559]: /org/bluez/hci0/dev_D4_E6_B7_DC_F8_E0/fd0: fd(24) ready

morrolinux, 2019-09-11:
You really should try Debian 9 instead of 10.
Also try `git checkout no_buffer_to_output` branch  and see if it works

squadramunter, 2019-09-11:
Can you provide an preinstalled image with this source code like you did with the older one? Maybe I do something wrong why it is not working?

morrolinux, 2019-09-11:
I've got this image but it's quite old, you should probably `git pull` and run the installer again:

https://mega.nz/#!pxpDkKQa!jgPfXpZHYArl-SkJizZTDKw7KWc9yLc09ld4K9pd9rk

squadramunter, 2019-09-11:
Sorry but I tested your image on Raspberry Pi 3B and it does not stream any FM signal to my Radio. Bluetooth connects but nothing more then that. It seems for now that your old repo works fine for me. mpradio seems to be more stable at the moment.

morrolinux, 2019-09-11:
@squadramunter You don't hear anything on FM because I gave you the Pi Zero W image (look at the filename). If I knew you had a Pi 3B I could have uploaded the correct image, but you didn't specify that and this issue is called "no sound from Bluetooth on **Raspberry Pi Zero W**". 
PiFmAdv needs to be recompiled across different models, so you might just solve it by simply:
1) `rm -rf /usr/local/src/PiFmAdv`
2) move to mpradio-py folder and `git pull` to update
3) run the installation process again

morrolinux, 2019-09-24:
Can't reproduce anymore, I'm gonna close this issue.