morrolinux/mpradio-py #45: Unable to run on boot

KlonoaOfTheWind, 2019-08-12:
I've just installed this today, to mess around with. But it doesn't start when the Pi boots. (Pi Zero W to be specific, running a light version of Raspbian Buster)

This is what I get when viewing the status of the service:
```
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-08-12 16:43:25 MDT; 32s ago
  Process: 440 ExecStart=/bin/bash -l -c /home/pi/mpradio.py (code=exited, status=1/FAILURE)
 Main PID: 440 (code=exited, status=1/FAILURE)

Aug 12 16:43:24 raspberrypi bash[440]: /etc/profile: line 4: id: command not found
Aug 12 16:43:24 raspberrypi bash[440]: /etc/profile: line 4: [: : integer expression expected
Aug 12 16:43:25 raspberrypi bash[440]: Traceback (most recent call last):
Aug 12 16:43:25 raspberrypi bash[440]:   File "/home/pi/mpradio.py", line 8, in <module>
Aug 12 16:43:25 raspberrypi bash[440]:     from bluetooth_remote import BtRemote
Aug 12 16:43:25 raspberrypi bash[440]:   File "/home/pi/mpradio/src/bluetooth_remote.py", line 2, in <module>
Aug 12 16:43:25 raspberrypi bash[440]:     import bluetooth
Aug 12 16:43:25 raspberrypi bash[440]: ModuleNotFoundError: No module named 'bluetooth'
Aug 12 16:43:25 raspberrypi systemd[1]: mpradio.service: Main process exited, code=exited, status=1/FAILURE
Aug 12 16:43:25 raspberrypi systemd[1]: mpradio.service: Failed with result 'exit-code'.

```

Any idea to what's going on?

morrolinux, 2019-08-13:
> ModuleNotFoundError: No module named 'bluetooth'

I think you're missing a dependency here. Awkward since the installation script should take care of this. Maybe the installer fails to install something on Buster? some dependencies could have change, I think you should try running the installation again and post your install log here

KlonoaOfTheWind, 2019-08-13:
Ran that installer again. Was seeing alot of "failed to fetch (file/repository)" errors. Turned out i needed to update my repositories, which I thought was strange as I ran that update a few days ago. Oh well, guess I probably didn't need to open this issue if I just slowed down and looked at what the installer was outputting lmao. Anyways seems like things are installing as they should be, I'll follow up on if it works or not.

KlonoaOfTheWind, 2019-08-13:
Well, that was that. I'm getting a signal now. But while I have your attention, may I ask how to get it to play mp3 files in a folder? Also how do I change the frequency? Also how do I set the RDS up? I just don't see it anywhere on the readme page.

morrolinux, 2019-08-13:
Oh yeah my bad, I didn't copy that readme part from the legacy project page because it was going to change. But it didn't so far, so there you go: https://github.com/morrolinux/mpradio/blob/master/README.md
PS: RDS is enabled by default

KlonoaOfTheWind, 2019-08-13:
Well, I assume I've done something wrong now as it's now exiting as failed again.
```
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-08-13 03:26:02 MDT; 2min 41s ago
  Process: 466 ExecStart=/bin/bash -l -c /home/pi/mpradio.py (code=exited, status=1/FAILURE)
 Main PID: 466 (code=exited, status=1/FAILURE)

Aug 13 03:26:01 raspberrypi bash[466]:     self.player = StoragePlayer()
Aug 13 03:26:01 raspberrypi bash[466]:   File "/home/pi/mpradio/src/storage_player.py", line 30, in __init__
Aug 13 03:26:01 raspberrypi bash[466]:     self.__rds_updater = RdsUpdater()
Aug 13 03:26:01 raspberrypi bash[466]:   File "/home/pi/mpradio/src/rds.py", line 26, in __init__
Aug 13 03:26:01 raspberrypi bash[466]:     self.__interval = int(config.get_settings()["RDS"]["updateInterval"])
Aug 13 03:26:01 raspberrypi bash[466]:   File "/usr/lib/python3.7/configparser.py", line 958, in __getitem__
Aug 13 03:26:01 raspberrypi bash[466]:     raise KeyError(key)
Aug 13 03:26:01 raspberrypi bash[466]: KeyError: 'RDS'
Aug 13 03:26:02 raspberrypi systemd[1]: mpradio.service: Main process exited, code=exited, status=1/FAILURE
Aug 13 03:26:02 raspberrypi systemd[1]: mpradio.service: Failed with result 'exit-code'.
```

My config file looks like this. I just copied the example and changed the frequency, and the station name.
```
[PIRATERADIO]
frequency=95.9
btGain=1.7
storageGain=1
treble=-6
output=fm
btBoost=false
implementation=pi_fm_adv

[PLAYLIST]
persistentPlaylist=true
resumePlayback=true
shuffle=true 

[RDS]
updateInterval=3
charsJump=6
stationName=Klonoa
rdsPattern=$ARTIST_NAME - $SONG_NAME

```
If I unplug the USB stick and reboot, it runs once more on the preset frequency, so I'm really certain it's something that I have done wrong.

Also I just want to ask, but would I just drop music onto the root directory of the USB stick? Or do I need to make a specific folder to put it in? Or am I going about this wrong?

KlonoaOfTheWind, 2019-08-13:
Yup, that's what I needed! Many thanks man. Also thank you for putting up with me, hehe.
You've got an awesome project though, and it seems like more stable than the PiFmAdv project by itself, which seemed to keep popping and cutting out for some reason.

closed 2019-08-13 by KlonoaOfTheWind

morrolinux, 2019-08-14:
I see that just now. Many thanks! But I don't really know how to compare the two, as mine is more of a wrapper for PiFmAdv

KlonoaOfTheWind, 2019-08-14:
Yours just seems to handle things better I guess. I've just had more trouble with piping audio into pifmadv and keeping it all alive.