morrolinux/mpradio-py #46: Possible Memory Leak?

KlonoaOfTheWind, 2019-08-14:
I can leave the program running for about an hour or so, before the system completely hangs and I have to pull the plug. Once it hangs the audio just stutters at the same part over and over again. When I checked htop, I found my swap completely full and my memory usuage was pretty much all used up. Most of the memory is used by the python script.
[image]

When recently rebooted, it takes about a third of the memory.
[image]

Currently the only solution that I would have is having a crontab reboot the system every hour or so.

KlonoaOfTheWind, 2019-08-14:
Also not quite relevant to this issue but once the program plays through all the files on the drive, it just stops playing. It won't pick any tracks all over again, unless I delete the library, playlist, and resume files. I'm not sure if that's normal or not tbh.

Also the RDS displays the song that first starts playing when it boots. But it doesn't scroll or update afterwards.

morrolinux, 2019-08-14:
Thanks for pointing that out. I've noticed this the other week or so on a long trip, but never really had the time to look into it and I eventually forgot. At least this will be a reminder.

Once the tracks are all played, the Pi should start a media scan over again, to make sure to include new added songs if any. This could take a while, but definitely I've always seen it working so I can't confirm, maybe investigate deeper into this and let me know?

RDS... Well this too is tested working but maybe I know why it doesn't for you: each car radio receiver has a different "supported" refresh rate. If you update more quickly than that, it will simply discard the update. I think it's what's happening now, you should try raising the update time in the configuration file to 5 seconds or more

morrolinux, 2019-08-14:
Alright so, I've noticed I wasn't calling input_container.close() upon each song's end, so that might be the issue. I pushed it on master, can you update and test it?

KlonoaOfTheWind, 2019-08-14:
Alright, I'll rerun the installer and follow up on the results.

And maybe. I'm on an iPod Nano 6th gen, it seems to update every few seconds with other FM stations that have RDS. So idk if it's something with the iPod or the program.

I dunno but whenever it's done playing everything it just plays silence. Like I said, if I delete those files then it rebuilds the library and then plays as normal.

morrolinux, 2019-08-14:
> Alright, I'll rerun the installer and follow up on the results.

obv you have to git pull before re-runnig the installer right?

> And maybe. I'm on an iPod Nano 6th gen, it seems to update every few seconds with other FM stations that have RDS. So idk if it's something with the iPod or the program.

Only way to know is by testing it. 3 seconds is pretty fast for most FM receivers. In your config file,
```
[RDS]
updateInterval=3          <-- change this to... say 8 seconds and see what happens
charsJump=6
stationName=MPRadio
rdsPattern=$ARTIST_NAME - $SONG_NAME

```

> I dunno but whenever it's done playing everything it just plays silence. Like I said, if I delete those files then it rebuilds the library and then plays as normal.

For how long does it play silence? can you stay attached with a shell and post some logs?

KlonoaOfTheWind, 2019-08-14:
Yeah, I pulled the new update. 

Also I'll give it a whack. Though I do notice that the ipod does update the RDS data from other stations every 3 to 4 seconds. 

And for the most part it'll play silence indefinitely. Though with a reinstall I'll see if any behavior changes.

KlonoaOfTheWind, 2019-08-14:
To update on the RDS, I've set the interval to 8 and it still won't scroll or change with whatever track is playing.

KlonoaOfTheWind, 2019-08-15:
Regarding the memory leak, it does not appear to be fixed. I left it running overnight and I tune in next morning to find the Pi locked up again. Watching htop from initial boot, the memory usage continues to climb, which it only seems to do so while it's loading a song. But it does not drop once a song has finished playing, it just continues to climb as the next song loads in.

morrolinux, 2019-08-15:
Alright so it turns out that pyAv has no "close()" attribute for input_container (but it does for output_container) so the only way to free the memory is to delete the object. seem to be working fine, try it out when you can

KlonoaOfTheWind, 2019-08-15:
I've been sorta watching the memory usage, It still seems to just be climbing. Once more, pulled the changes and then reran the installer. Even deleted the entire cloned repo and recloned, and tried installing again. I'll let it run but I feel like it's going to just lock up again after a bit.

morrolinux, 2019-08-15:
It's normal for it to climb during playback since a data structure is being filled, the important thing is for it to stop growing "after the first song"

KlonoaOfTheWind, 2019-08-15:
Unfortunately it doesn't seem to stop growing. Several songs have played through, and it just continues to eat memory.

KlonoaOfTheWind, 2019-08-15:
Just a thought, but what if it's just not unloading the music after it has played? I wonder because initially, the memory usuage climbs at a pretty high rate (most likely the mentioned data structure). But with each song that plays, it climbs more slowly, usually depending on the size of the track.

KlonoaOfTheWind, 2019-08-16:
I have let it run to return and find the system locked up again. As sort of a bandaid fix I just have a bash script restarting the service every 30 mins or so, but I guess it sometimes also doesn't keep up cause I at one point ended up having to restart the pi again.

Also I wonder if the system locking up like that sometimes causes issues with resuming media, as when I restarted the pi, it finished a track but didn't start playing any others until i again deleted those library files.

morrolinux, 2019-08-16:
I'm sorry to hear that...
I think playlist and library files could get corrupted when you forcibly shut it off after the lockdown. 
Do you know a bit of Python? I'm going to have holidays and will be unavailable for about a week, but if I were in you I would try to force a call to `gc.collect` (python garbage collector) right after the deletion of `input_container` in `storage_player.py`

KlonoaOfTheWind, 2019-08-16:
Eh, not your fault, you got nothin to be sorry about. I just think though that might've been to why it wouldn't play music sometimes.

Also I honestly don't. Though I could probably read up a bit on it. Hell, I'd probably better off if I actually got into this stuff more... 
But it doesn't seem to hard to add garbage collection to it, so maybe I'll give that a swing

KlonoaOfTheWind, 2019-08-17:
Just calling `gc.collect` just causes the program to play one song then stop playing. So I assume it's removing something that's important from memory when it's called.

morrolinux, 2019-09-13:
Something awkward just happened. 
After around 1:20 hours of music playback, I've got it to underrun. 
I could hear the last 500 ms of sound being repeated over and over, but I was connected to the Pi 3 and able to launch `htop`. 
Here's what's awkward: 
[image]
I'd say memory usage is more or less stable and definitely not causing the issue, being far from the sturation point. 
I think with this ticket you pointed out a memory leak issue which then got solved but the actual reason for the underrun might be somewhere else.

KlonoaOfTheWind, 2019-09-13:
I looked at the PiFmAdv program, there's mention of buffer underruns. Maybe this is an issue with that part, and not just necessarily your program.

closed 2019-09-24 by morrolinux