morrolinux/mpradio-py #49: Usb stick and controlls

vvervolv, 2019-09-23:
Good afternoon! 
Thank you so much for Mpradio-py! But I have some problems.
1. I use Raspberry PI 3 B+. After installation, I succesful connect to the device via bluetooth, but unfortunately the music was not broadcast on FM.
But, if I connect a USB stick with a configuration file, everything is OK (Except for the companion application, it does not allow doing anything). I want to use my PI in the car, so I want to minimize the size, so how can i make it works without usb stick?
2. My push-button control does not work (GPIO-connected on pin 5 [BCM 3]).

I guess that GPIO and application problems are same 

Please, tell me where to look for an error

P.S. Sorry for my English

morrolinux, 2019-09-23:
Hi, 
Can you elaborate on 1.) describing what happens when you try to connect via Bluetooth for audio playback? 
Right now during `mpradio-py` installation a line is written in `/etc/fstab` for automatically mounting a device called `/dev/sda1` (which would be your USB stick if you plug it in). You could partition your SD CARD making a new `FAT32` partition and storing everything there, but to make it work in this configuration, you should change `/dev/sda1` with the partition name you just created. you can find it using `sudo fdisk -l` looking for the last partition.

as for 2.) can you test by simply short-circuiting with GND? I'm gonna guess it's an hardware problem unless an exception is raised in the logs of `journalctl -f mpradio`

vvervolv, 2019-09-24:
Hello!
Thank you for answer! 
When i try to connect my phone to Pi (without usb stick),  it connects succesful (`bluetoothctl` says that). But when i try to play music, only silence (not noise)  is heard from the radio (and MPRadio remote app does not start, like PI is disconnect). 
`sudo systemctl status mpradio` says : 

> pi@raspberrypi:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-09-24 09:33:13 MSK; 5min ago
 Main PID: 459 (python3)
    Tasks: 11 (limit: 2200)
   Memory: 57.7M
   CGroup: /system.slice/mpradio.service
           ├─459 python3 /home/pi/mpradio.py
           ├─540 sox -t raw -G -b 16 -e signed -c 2 -r 44100 - -t wav - compand 0.3,1 6:-70,-60,-20 -5 -90 0.2 treble -6
           ├─544 sudo pi_fm_adv --freq 88.0 --ctl /tmp/rds_ctl --audio -
           └─551 pi_fm_adv --freq 88.0 --ctl /tmp/rds_ctl --audio -
Sep 24 09:33:13 raspberrypi systemd[1]: Started Morrolinux Pirate Radio.
Sep 24 09:33:18 raspberrypi bash[459]: /etc/profile: line 4: id: command not found
Sep 24 09:33:18 raspberrypi bash[459]: /etc/profile: line 4: [: : integer expression expected
Sep 24 09:33:21 raspberrypi sudo[544]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/local/bin/pi_fm_adv --freq 88.0 --ctl /tmp/rds_ctl --audio -
Sep 24 09:33:21 raspberrypi sudo[544]: pam_unix(sudo:session): session opened for user root by (uid=0)

But when i plug in usb stick,  `integer expression expected` dissapeared, and it starts  play music from usb and from bluetooth (App starts, but buttons (like halt or reboot) not working). 

I will try to make FAT32 partition tommorow, i forgot my cardreader at home :)

For 2.)
I try to short-circuiting with GND, but nothing happens. When i try to `journalctl -f mpradio` it says 

> Failed to add match 'mpradio': Invalid argument

But it works on previous version of MPRadio (the old implementation)
Is it possible that the problem is caused by the NOOBS->raspberry lite distributive?

Thank you!

morrolinux, 2019-09-24:
> But when i plug in usb stick, `integer expression expected` dissapeared

That error is not relevant to `mpradio`

> and it starts play music from usb and from bluetooth 

By that you mean you're then able to play bluetooth audio from your phone as well? 

> (App starts, but buttons (like halt or reboot) not working).

The App is being adapted from the legacy `mpradio` version server backend. There are some minor differences and I haven't completed it yet. What should work: 
1. Tracks listing and on demand track playback
2. Skip button
3. Maybe some others but I'm not sure

> I try to short-circuiting with GND, but nothing happens. When i try to `journalctl -f mpradio` it says

Oh yeah sorry about that it's `journalctl -fu mpradio`. If you don't specify `u` you're not telling `journalctl` which unit to look at.

> Is it possible that the problem is caused by the NOOBS->raspbian lite distributive?

Speaking of Debian... which version are you running on? 9 or 10?

morrolinux, 2020-03-08:
closing for inactivity

closed 2020-03-08 by morrolinux

GiannisDorokidis, 2020-07-25:
How can I make a new partition?