morrolinux/mpradio-py #50: RPi 3 GPIO 5 has a pull-up resistor, constantly firing "poweroff"

96Octavian, 2019-10-01:
With a clean install of Raspbian Buster for the Rasperry Pi 3 mpradio works just out of the box but whenever I connect a bluetooth devices it receives a "poweroff" command and shuts down.
I don't know why it receives it only after a bluetooth connection, but I logged around and found out that at line 140 of https://github.com/morrolinux/mpradio-py/blob/master/src/gpio_remote.py it was receiving an input from GPIO, self.remote_msg contained 
`{'command': ['system', 'poweroff'], 'source': 'gpio'}`
Also when starting `mpradio.py` it gave me this warning:
`/home/pi/mpradio/src/gpio_remote.py:36: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(5, GPIO.IN, pull_up_down=GPIO.PUD_UP)`
I think it's because GPIO5 in Pi 3 is always HIGH so it constantly triggers, maybe? I tried using GPIO6 but I get this error instead:
`Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/mpradio/src/gpio_remote.py", line 36, in __run
    GPIO.setup(6, GPIO.IN, pull_up_down=GPIO.PUD_UP)
ValueError: The channel sent is invalid on a Raspberry Pi`
Although this lets me play both FM and bluethoot just fine

morrolinux, 2019-10-01:
I always felt it could cause trouble but eventually I forgot to fix that...
You should try to remove `pull_up_down=GPIO.PUD_UP` and see how it works.
Probably this will fix it, please let me know

96Octavian, 2019-10-03:
I tried your suggestion and also tried using other pins but it didn't work, I found out that in [https://github.com/morrolinux/mpradio-py/blob/f7ed8df7aef658c755701c2d74d57ee67f2ec4bc/src/gpio_remote.py#L45](url) if I write 
`if input_state == GPIO.HIGH:`
instead, the event is not fired anymore, but I'm yet to add a button to test if it works

morrolinux, 2019-10-05:
Ok let me know!

closed 2020-03-08 by morrolinux