morrolinux/mpradio-py #51: Android app keeps crashing

RichMStudios, 2019-11-17:
Hello,
I installed the app on my Android 9 device. It asks for storage permissions and then crashes.
I know you mentioned it can crash (since it's the pre-alpha build), but I'm just interested if you released a newer version or something.
Thanks in advance!

morrolinux, 2019-11-17:
Try [this](https://www.mediafire.com/file/h4tpihsd0y1lziv/mpradio-py_0.2.apk/file) build (it's my latest and it's running "fine" on my Android 9 based device so it should work for you as well as long as:
1) you have your bluetooth on
2) the Pi is within reach
3) you previously paired the pi via android settings
4) your pi is showing up as "mpradio" in your android bluetooth settings

And remember that the app is only useful if you want to change some settings or control playback from usb storage. 
You don't need the app for bluetooth music streaming such as spotify, youtube, ecc..

kiam001, 2019-12-21:
hello morrolinux the app is working on my android 9 but only whan the pi is was connectet

morrolinux, 2019-12-21:
> hello morrolinux the app is working on my android 9 but only whan the pi is was connectet

that's "normal" since I haven't handled corner conditions yet. For the moment the most important thing is that the latest version of the app mostly works with the latest version of mpradio-py

closed 2020-03-08 by morrolinux