morrolinux/mpradio-py #52: write in readme what debian version is requestet on debian brusters dont work

kiam001, 2019-12-21:


kiam001, 2019-12-21:
okay stretch works all not

kiam001, 2019-12-21:
give as a pre installed image?

kiam001, 2019-12-21:
mp3 played but whan i connect my nokia 6.1 via bluetooth plays the mp3 and no bluetooth sound

morrolinux, 2019-12-21:
can you provide bluetooth and mpradio logs?
You find info about how to get those in the readme

kiam001, 2019-12-22:
pi@mpradio:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-12-22 08:47:18 GMT; 4min 56s ago
 Main PID: 557 (sudo)
    Tasks: 13 (limit: 2200)
   Memory: 75.9M
   CGroup: /system.slice/mpradio.service
           ├─557 sudo -E chrt -f 99 /home/pi/mpradio.py
           ├─614 python3 /home/pi/mpradio.py
           ├─670 sox -t raw -G -b 16 -e signed -c 2 -r 44100 - -t wav - compand 0.3,1 6:-70,-60,-20 -5 -90 0.2 treble -6
           ├─671 sudo pi_fm_adv --freq 108.0 --ctl /tmp/rds_ctl --audio -
           └─682 pi_fm_adv --freq 108.0 --ctl /tmp/rds_ctl --audio -

Dec 22 08:47:29 mpradio bash[557]: Error reading lyrics, skipped
Dec 22 08:47:29 mpradio bash[557]: Estimating duration from bitrate, this may be inaccurate
Dec 22 08:47:29 mpradio bash[557]: Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Dec 22 08:47:29 mpradio bash[557]: Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the times
Dec 22 08:47:29 mpradio bash[557]: Encoder did not produce proper pts, making some up.
Dec 22 08:50:33 mpradio bash[557]: Incorrect BOM value
Dec 22 08:50:33 mpradio bash[557]: Error reading lyrics, skipped
Dec 22 08:50:33 mpradio bash[557]: Estimating duration from bitrate, this may be inaccurate
Dec 22 08:50:33 mpradio bash[557]: Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Dec 22 08:50:33 mpradio bash[557]: Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the times
lines 1-23/23 (END)

kiam001, 2019-12-22:
```
pi@mpradio:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-12-22 08:47:18 GMT; 9min ago
 Main PID: 557 (sudo)
    Tasks: 13 (limit: 2200)
   Memory: 86.4M
   CGroup: /system.slice/mpradio.service
           ├─557 sudo -E chrt -f 99 /home/pi/mpradio.py
           ├─614 python3 /home/pi/mpradio.py
           ├─670 sox -t raw -G -b 16 -e signed -c 2 -r 44100 - -t wav - compand 0.3,1 6:-70,-60,-20 -5 -90 0.2 treble -6
           ├─671 sudo pi_fm_adv --freq 108.0 --ctl /tmp/rds_ctl --audio -
           └─682 pi_fm_adv --freq 108.0 --ctl /tmp/rds_ctl --audio -

Dec 22 08:53:35 mpradio bash[557]: Incorrect BOM value
Dec 22 08:53:35 mpradio bash[557]: Error reading lyrics, skipped
Dec 22 08:53:35 mpradio bash[557]: Estimating duration from bitrate, this may be inaccurate
Dec 22 08:53:35 mpradio bash[557]: Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Dec 22 08:53:35 mpradio bash[557]: Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the times
Dec 22 08:56:46 mpradio bash[557]: Incorrect BOM value
Dec 22 08:56:46 mpradio bash[557]: Error reading lyrics, skipped
Dec 22 08:56:46 mpradio bash[557]: Estimating duration from bitrate, this may be inaccurate
Dec 22 08:56:46 mpradio bash[557]: Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Dec 22 08:56:46 mpradio bash[557]: Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the times
lines 1-23/23 (END)

```

```
pi@mpradio:~ $ cat /var/log/bluetooth_dev
[Sun Dec 22 08:52:05 GMT 2019]: Start Bluetooth played connection  via python script
pi@mpradio:~ $

```
```
pi@mpradio:~ $ hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:AF:65:CA  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN ISCAN
        RX bytes:5364 acl:92 sco:0 events:134 errors:0
        TX bytes:5794 acl:93 sco:0 commands:58 errors:0

pi@mpradio:~ $

```
```
pi@mpradio:~ $ sudo systemctl status mpradio-bt-setup
● mpradio-bt-setup.service - Mpradio bluetooth setup
   Loaded: loaded (/etc/systemd/system/mpradio-bt-setup.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2019-12-22 08:47:20 GMT; 11min ago
  Process: 336 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 563 ExecStart=/bin/bash -l -c /usr/local/bin/mpradio-bt-setup.sh (code=exited, status=0/SUCCESS)
 Main PID: 563 (code=exited, status=0/SUCCESS)

Dec 22 08:47:09 mpradio systemd[1]: Starting Mpradio bluetooth setup...
Dec 22 08:47:20 mpradio bash[563]: Serial Port service registered
Dec 22 08:47:20 mpradio systemd[1]: Started Mpradio bluetooth setup.
pi@mpradio:~ $

```
```
pi@mpradio:~ $ sudo systemctl status simple-agent
● simple-agent.service - Bluealsa audio server
   Loaded: loaded (/etc/systemd/system/simple-agent.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-12-22 08:47:20 GMT; 12min ago
 Main PID: 579 (simple-agent)
    Tasks: 1 (limit: 2200)
   Memory: 8.5M
   CGroup: /system.slice/simple-agent.service
           └─579 /usr/bin/python /usr/local/bin/simple-agent

Dec 22 08:47:20 mpradio systemd[1]: Started Bluealsa audio server.
pi@mpradio:~ $

```
```
pi@mpradio:~ $  sudo systemctl status bluealsa
● bluealsa.service - Bluealsa audio server
   Loaded: loaded (/etc/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-12-22 08:47:20 GMT; 13min ago
 Main PID: 580 (bluealsa)
    Tasks: 3 (limit: 2200)
   Memory: 3.0M
   CGroup: /system.slice/bluealsa.service
           └─580 /usr/bin/bluealsa -p a2dp-sink --a2dp-force-audio-cd

Dec 22 08:47:20 mpradio systemd[1]: Started Bluealsa audio server.
pi@mpradio:~ $

```

morrolinux, 2019-12-22:
Ok those logs are not really useful, maybe you can try this:
1) systemctl disable mpradio
2) reboot
3) ssh into your pi
4) sudo ./mpradio.py   (this wil print all program output to terminal)
5) once you hear the music playing from storage, try connecting your phone via bluetooth 
6) paste here the shell output

kiam001, 2019-12-25:
pi@mpradio:~ $ sudo ./mpradio.py
mpradio main PID is: 617
broadcasting on FM 108.0
storage_player playing: /pirateradio/The\ Angels\ Sang\ -\ REBA\ MCENTIRE.mp3
Exception in thread Thread-8:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "./mpradio.py", line 123, in check_remotes
    tmp = BtPlayerLite(cmd[2])
IndexError: list index out of range

Estimating duration from bitrate, this may be inaccurate
Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStr          eam.codecpar instead.
Timestamps are unset in a packet for stream 0. This is deprecated and will stop           working in the future. Fix your code to set the timestamps properly
Encoder did not produce proper pts, making some up.

kiam001, 2019-12-25:
and the bluetooth interface can not find on mobile

kiam001, 2019-12-25:
the rds is buggy too

the old image mpradio-multi1.1 work perfekt only remote frequnzset not working

kiam001, 2019-12-25:
after strg + c

```
^Cstopping threads and clean termination...
transcoding finished.
Exception ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
  File "./mpradio.py", line 69, in termination_handler
    quit(0)
  File "/usr/lib/python3.7/_sitebuiltins.py", line 26, in __call__
    raise SystemExit(code)
SystemExit: 0
pi@mpradio:~ $

```

kiam001, 2019-12-25:
the newst buster image work but after a full upgrade it shutdowns on evry start

on jessie after install go it in shutdown mode

kiam001, 2019-12-25:
i hope you can the issus fixed or deploy a working readonly image

kiam001, 2019-12-26:
mybe i have found a soultion https://www.sigmdel.ca/michel/ha/rpi/bluetooth_n_buster_01_en.html

morrolinux, 2019-12-26:
did it work?

kiam001, 2019-12-26:
no :(

kiam001, 2019-12-26:
bluetooth is displayed as "mpradio" whan i connect my phones says connecected and drop the connection after 10 secounds 

the music plays mp3 all time

morrolinux, 2019-12-28:
I've got this image but it's quite old, you should probably git pull and run the installer again:

https://mega.nz/#!pxpDkKQa!jgPfXpZHYArl-SkJizZTDKw7KWc9yLc09ld4K9pd9rk

closed 2020-02-04 by morrolinux