morrolinux/mpradio-py #53: bluetooth app and audio passthrough to fm doesnt work

GiannisDorokidis, 2020-01-29:
Hello I am having issues with Bluetooth not passing the audio to the pi (zero w).
I can connect to pi's Bluetooth but the app doesn't work. 
I'm tring to passthrough my phones audio but it doesn't work. 
The usb stick works perfectly and it is trasmitting music from the usb stick. 
Is there any way to fix this?

morrolinux, 2020-01-29:
which debian/raspbian version are you running on? Another user recently reported issues with bluetooth playback under Debian 10 (Buster). 
This project was developed on Debian Debian 9 (Stretch) and I hadn't got time to test it under Buster yet, so that might simply be the reason. Could you try flashing raspbian 9 and install again?

GiannisDorokidis, 2020-01-29:
I gloogled to find this 
pi@mpradio:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:Raspbian
Description:Raspbian GNU/Linux 10 (buster)
Release:10
Codename:buster
pi@mpradio:~
And then I did this
Linux mpradio 4.19.93+ #1290 Fri Jan 10 16:34:37 GMT 2020 armv6l GNU/Linux

GiannisDorokidis, 2020-01-29:
were can I find raspbian stretch? Can I download it from NOOBS? 
P.S I New to this stuff

morrolinux, 2020-01-29:
http://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/

GiannisDorokidis, 2020-01-29:
I tried to run it on raspbian stretch but it has failed
pi@mpradio:~ $ sudo systemctl status mpradio
● mpradio.service - Morrolinux Pirate Radio
   Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor pr   Active: failed (Result: exit-code) since Wed 2020-01-29 22:07:06 GMT; 5  Process: 334 ExecStart=/bin/bash -l -c sudo -E chrt -f 99 /home/pi/mprad Main PID: 334 (code=exited, status=1/FAILURE)

● mpradio.service - Morrolinux Pirate Radio                                  Loaded: loaded (/etc/systemd/system/mpradio.service; enabled; vendor pr   Active: failed (Result: exit-code) since Wed 2020-01-29 22:07:06 GMT; 5  Process: 334 ExecStart=/bin/bash -l -c sudo -E chrt -f 99 /home/pi/mprad Main PID: 334 (code=exited, status=1/FAILURE)                                                                                                      Jan 29 22:07:06 mpradio bash[334]: Traceback (most recent call last):     Jan 29 22:07:06 mpradio bash[334]:   File "/home/pi/mpradio.py", line 8, iJan 29 22:07:06 mpradio bash[334]:     from bluetooth_remote import BtRemoJan 29 22:07:06 mpradio bash[334]:   File "/home/pi/mpradio/src/bluetooth_Jan 29 22:07:06 mpradio bash[334]:     import bluetooth                   Jan 29 22:07:06 mpradio bash[334]: ImportError: No module named 'bluetoothJan 29 22:07:06 mpradio sudo[334]: pam_unix(sudo:session): session closed Jan 29 22:07:06 mpradio systemd[1]: mpradio.service: Main process exited, Jan 29 22:07:06 mpradio systemd[1]: mpradio.service: Unit entered failed sJan 29 22:07:06 mpradio systemd[1]: mpradio.service: Failed with result 'e

morrolinux, 2020-01-30:
looks like `pybluez` package didn't get installed... which is awkward because the installer script should take care of it.
1) Did you run }install.sh} with `sudo` following the README?
2) what happens if you type `pip3 install pybluez`?

GiannisDorokidis, 2020-01-30:
pi@mpradio:~ $ ls
Desktop    Downloads  mpradio     mpradio.py  Pictures  Templates
Documents  MagPi      mpradio-py  Music       Public    Videos
pi@mpradio:~ $ cd mpradio
pi@mpradio:~/mpradio $ ls
a2dp-sink.py  doc      mpradio-py-notes.txt  rfcomm-client.py  sounds
classes.png   install  README.md             songs             src
pi@mpradio:~/mpradio $ sudo nano README.md
pi@mpradio:~/mpradio $ cd mpradio/install && sudo bash install.sh
-bash: cd: mpradio/install: No such file or directory
pi@mpradio:~/mpradio $ sudo bash install.sh                               bash: install.sh: No such file or directory
pi@mpradio:~/mpradio $ cd mpradio.py                                      -bash: cd: mpradio.py: No such file or directory
pi@mpradio:~/mpradio $ pip3 install pybluez                               
Collecting pybluez
  Downloading https://files.pythonhosted.org/packages/08/9f/e9d93b266d2d1ea988780a52a696073ba0a65df65a532165fdf6ff90d0ed/PyBluez-0.23.tar.gz (97kB)
    100% |████████████████████████████████| 102kB 301kB/s                 Building wheels for collected packages: pybluez
  Running setup.py bdist_wheel for pybluez ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-13fyw4pa/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpi0_lp0pppip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv6l-3.5
  creating build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/ble.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/widcomm.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/__init__.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/macos.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/btcommon.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/bluez.py -> build/lib.linux-armv6l-3.5/bluetooth
  copying bluetooth/msbt.py -> build/lib.linux-armv6l-3.5/bluetooth
  Fixing build/lib.linux-armv6l-3.5/bluetooth/ble.py build/lib.linux-armv6l-3.5/bluetooth/widcomm.py build/lib.linux-armv6l-3.5/bluetooth/__init__.py build/lib.linux-armv6l-3.5/bluetooth/macos.py build/lib.linux-armv6l-3.5/bluetooth/btcommon.py build/lib.linux-armv6l-3.5/bluetooth/bluez.py build/lib.linux-armv6l-3.5/bluetooth/msbt.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  Fixing build/lib.linux-armv6l-3.5/bluetooth/ble.py build/lib.linux-armv6l-3.5/bluetooth/widcomm.py build/lib.linux-armv6l-3.5/bluetooth/__init__.py build/lib.linux-armv6l-3.5/bluetooth/macos.py build/lib.linux-armv6l-3.5/bluetooth/btcommon.py build/lib.linux-armv6l-3.5/bluetooth/bluez.py build/lib.linux-armv6l-3.5/bluetooth/msbt.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'bluetooth._bluetooth' extension
  creating build/temp.linux-armv6l-3.5
  creating build/temp.linux-armv6l-3.5/bluez
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I./port3 -I/usr/include/python3.5m -c bluez/btmodule.c -o build/temp.linux-armv6l-3.5/bluez/btmodule.o
  In file included from bluez/btmodule.c:20:0:
  bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No such file or directory
   #include <bluetooth/bluetooth.h>
                                   ^
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pybluez
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez
  Running setup.py install for pybluez ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-13fyw4pa/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9ggjzrrw-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv6l-3.5
    creating build/lib.linux-armv6l-3.5/bluetooth
    copying bluetooth/ble.py -> build/lib.linux-armv6l-3.5/bluetooth
    copying bluetooth/widcomm.py -> build/lib.linux-armv6l-3.5/bluetooth
    copying bluetooth/__init__.py -> build/lib.linux-armv6l-3.5/bluetooth
    copying bluetooth/macos.py -> build/lib.linux-armv6l-3.5/bluetooth
[... 3 lines cut ...]
    Fixing build/lib.linux-armv6l-3.5/bluetooth/ble.py build/lib.linux-armv6l-3.5/bluetooth/widcomm.py build/lib.linux-armv6l-3.5/bluetooth/__init__.py build/lib.linux-armv6l-3.5/bluetooth/macos.py build/lib.linux-armv6l-3.5/bluetooth/btcommon.py build/lib.linux-armv6l-3.5/bluetooth/bluez.py build/lib.linux-armv6l-3.5/bluetooth/msbt.py
    Skipping optional fixer: buffer
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
[... cut ...]
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-13fyw4pa/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9ggjzrrw-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-13fyw4pa/pybluez/
pi@mpradio:~/mpradio $

GiannisDorokidis, 2020-01-30:
should I try to re-install raspian stretch then downloading the Project?

GiannisDorokidis, 2020-01-30:
pi@mpradio:~ $ sudo systemctl
  UNIT                      LOAD   ACTIVE SUB       DESCRIPTION             proc-sys-fs-binfmt_misc.automount loaded active waiting   Arbitrary Exec  sys-devices-platform-soc-20201000.serial-tty-ttyAMA0-hci0.device loaded   sys-devices-platform-soc-20201000.serial-tty-ttyAMA0.device loaded activ  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20300000.mmcnr-mmc_host-mmc1-mmc1:0001-mmc1:000  sys-devices-platform-soc-20980000.usb-usb1-1\x2d1-1\x2d1:1.0-host0-targe  sys-devices-platform-soc-20980000.usb-usb1-1\x2d1-1\x2d1:1.0-host0-targe  sys-devices-platform-soc-soc:audio-sound-card0.device loaded active plug  sys-devices-virtual-block-ram0.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram1.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram10.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram11.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram12.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram13.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram14.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram15.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram2.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram3.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram4.device loaded active plugged   /sys/devic  UNIT                      LOAD   ACTIVE SUB       DESCRIPTION             proc-sys-fs-binfmt_misc.automount loaded active waiting   Arbitrary Exec  sys-devices-platform-soc-20201000.serial-tty-ttyAMA0-hci0.device loaded   sys-devices-platform-soc-20201000.serial-tty-ttyAMA0.device loaded activ  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20202000.mmc-mmc_host-mmc0-mmc0:aaaa-block-mmcb  sys-devices-platform-soc-20300000.mmcnr-mmc_host-mmc1-mmc1:0001-mmc1:000  sys-devices-platform-soc-20980000.usb-usb1-1\x2d1-1\x2d1:1.0-host0-targe  sys-devices-platform-soc-20980000.usb-usb1-1\x2d1-1\x2d1:1.0-host0-targe  sys-devices-platform-soc-soc:audio-sound-card0.device loaded active plug  sys-devices-virtual-block-ram0.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram1.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram10.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram11.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram12.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram13.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram14.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram15.device loaded active plugged   /sys/devi  sys-devices-virtual-block-ram2.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram3.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram4.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram5.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram6.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram7.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram8.device loaded active plugged   /sys/devic  sys-devices-virtual-block-ram9.device loaded active plugged   /sys/devic  sys-devices-virtual-misc-rfkill.device loaded active plugged   /sys/devi  sys-devices-virtual-tty-ttyprintk.device loaded active plugged   /sys/de  sys-module-configfs.device loaded active plugged   /sys/module/configfs   sys-module-fuse.device    loaded active plugged   /sys/module/fuse        sys-subsystem-bluetooth-devices-hci0.device loaded active plugged   /sys  sys-subsystem-net-devices-wlan0.device loaded active plugged   /sys/subs  -.mount                   loaded active mounted   Root Mount              boot.mount                loaded active mounted   /boot                   dev-mqueue.mount          loaded active mounted   POSIX Message Queue Fi  pirateradio.mount         loaded active mounted   /pirateradio            run-rpc_pipefs.mount      loaded active mounted   RPC Pipe File System    run-user-1000-gvfs.mount  loaded active mounted   /run/user/1000/gvfs     run-user-1000.mount       loaded active mounted   /run/user/1000          sys-fs-fuse-connections.mount loaded active mounted   FUSE Control File   sys-kernel-config.mount   loaded active mounted   Configuration File Sys  sys-kernel-debug.mount    loaded active mounted   Debug File System       systemd-ask-password-plymouth.path loaded active waiting   Forward Passw  systemd-ask-password-wall.path loaded active waiting   Forward Password   init.scope                loaded active running   System and Service Man  session-c1.scope          loaded active running   Session c1 of user pi   session-c2.scope          loaded active running   Session c2 of user pi   session-c3.scope          loaded active running   Session c3 of user pi   alsa-restore.service      loaded active exited    Save/Restore Sound Car  avahi-daemon.service      loaded active running   Avahi mDNS/DNS-SD Stac  bluealsa.service          loaded active running   Bluealsa audio s [... cut ...]                                                             LOAD   = Reflects whether the unit definition was properly loaded.        ACTIVE = The high-level unit activation state, i.e. generalization of SUB.SUB    = The low-level unit activation state, values depend on unit type.                                                                           137 loaded units listed. Pass --all to see loaded but inactive units, too.To show all installed unit files use 'systemctl list-unit-files'

>  @

morrolinux, 2020-01-30:
`libbluetooth-dev` is also missing. And that too is being installed by the installer script. 
Please retry the install by doing exactly what the README says:
(you can copy-paste this for simplicity)

1) login to your Raspberry Pi
2) `cd mpradio/install && sudo bash install.sh`

Paste here the shell output. but please use pastebin.com if you manage

GiannisDorokidis, 2020-01-30:
re run the commant "cd mpradio/install && sudo bash install.sh''
Then this 
https://drive.google.com/file/d/18oUlMkOhEsljtwwwdgYmsiLt0vwak6D7/view?usp=drivesdk

morrolinux, 2020-01-30:
c'mon man first file is truncated and for the second I don't have access. Why didn't you use https://pastebin.com ?

morrolinux, 2020-01-30:
Anyway from what I can read on the first half it looks like a build issue with pyav library: https://github.com/mikeboers/PyAV
You should really go open a ticket there providing all the requested information which are needed to investigate on that

GiannisDorokidis, 2020-01-30:
sorry wasting your time here is the links:
https://pastebin.com/K52hWMEF
https://pastebin.com/f3MPRUvn

GiannisDorokidis, 2020-01-31:
I made a forum on pyAv and they told me what version of FFmpeg I have how can I find that?

morrolinux, 2020-01-31:
you can really easily google that: https://www.google.com/search?q=linux+which+version+of+ffmpeg+am+I+running&oq=linux+which+version+of+ffmpeg+am+I+running&aqs=chrome..69i57j69i64.7723j0j7&sourceid=chrome&ie=UTF-8
it's the first result.

Take this as a general approach

GiannisDorokidis, 2020-02-01:
They told me that pyAv isn't supported because they don't support FFmpeg v3.0 so thats why its not working on raspian stretch
Thank you for your time and work

morrolinux, 2020-02-01:
Ok I've read the issue on the pyAV. They switched to version 7 wich no longer supports ffmpeg < 4.
Lukily we can install a specific version via pip.
Try this command and let me know:
`pip install 'av==6'`

GiannisDorokidis, 2020-02-01:
I tried this and then this happened 
https://pastebin.com/614zKqWf
Sould I try to install av before mpradio it self ?

morrolinux, 2020-02-01:
I need to see the output of: `sudo pip install 'av==6'`

GiannisDorokidis, 2020-02-01:
here is pip install av==6
https://pastebin.com/RF6RK3Fw

morrolinux, 2020-02-01:
Since the av==6 installation goes well, I now updated the install script to install that instead of the latest version. try pulling the new version, run the installer and paste here both the installer log and the `systemctl status mpradio`

GiannisDorokidis, 2020-02-02:
installation https://pastebin.com/9vmurmH2
mpradio status https://pastebin.com/y0c3QQnX

morrolinux, 2020-02-02:
Looks like we moved one step ahead. Now the application is crashing on a routine it never really crashed before, is the config file (`pirateradio.config`) in the right place? what's its content?

GiannisDorokidis, 2020-02-02:
I had problem with pirateradio.config Now its streaming properly here is the status
https://pastebin.com/PgYnmctX
I will Now try the bluetooth
also RDS is working finaly properly

GiannisDorokidis, 2020-02-02:
bluetooth doesn't work its disconecting after 10 seconds and I can't controll the pi with the apk

morrolinux, 2020-02-02:
yeah bt-setup is legacy.

Is it streaming 10 seconds of audio and then disconnecting or does it stay connected for 10 seconds without any transmission working?
Even in this case I would need the mpradio log as always

GiannisDorokidis, 2020-02-02:
its streaming from the usb stick only I can connect to the pi via bluetooth with no audio and after 10 seconds it disconects
https://pastebin.com/Qaup2eK0

morrolinux, 2020-02-02:
what does `cat /var/log/bluetooth_dev` say?
is `mpradio.service` enabled?

try issuing `sudo systemctl stop mpradio && sudo rm /tmp/mpradio_bt && sudo systemctl enable mpradio && sudo reboot`

After the reboot, wait until you hear some music from storage, THEN try connecting your phone to your Pi's bluetooth like you would normally do for any bluetooth speaker and try playing something.

IF it doesn't work, log back in to your Pi and post again the status of mpradio.service

Let me know

GiannisDorokidis, 2020-02-02:
pi@mpradio:~ $ cat /var/log/bluetooth_dev                                 [Sun Feb  2 18:28:21 GMT 2020]: Start Bluetooth played connection  via python script
 
Then its says mpradio.service is running
After reboot the same
mpradio.service 
https://pastebin.com/CrgGJXyN

morrolinux, 2020-02-02:
looks like `a2dp_connected.py` script is unable to find the connected bluetooth device and report its mac address. Strange, I'll have to think of something to debug this

GiannisDorokidis, 2020-02-02:
I try again to run "sudo systemctl stop mpradio && sudo rm /tmp/mpradio_bt && sudo systemctl enable mpradio && sudo reboot"
And bluetooth is Now connected but still can't stream music over bluetooth

morrolinux, 2020-02-02:
try this:
1) on the console, write "python"
2) paste this:
```
from __future__ import absolute_import, print_function, unicode_literals
import dbus
bus = dbus.SystemBus()
manager = dbus.Interface(bus.get_object("org.bluez", "/"), "org.freedesktop.DBus.ObjectManager")
objects = manager.GetManagedObjects()

for path, interfaces in objects.items():
    print("PATH: ", path, "\n INTERFACES", interfaces, "\n KEYS", interfaces.keys())
    print("-----------------------------")
```
and attach the output as always

GiannisDorokidis, 2020-02-02:
output https://pastebin.com/vwMcBTm1

morrolinux, 2020-02-02:
what's the output of running on the shell `a2dp_connected.py` ?

GiannisDorokidis, 2020-02-02:
https://pastebin.com/3mk2GC9U
Is this what you wanted to see? because I didn't understand

morrolinux, 2020-02-02:
no man just type the above command on the bash shell

morrolinux, 2020-02-02:
(you'd have to quit the python shell you're in)

GiannisDorokidis, 2020-02-02:
pi@mpradio:~ $ a2dp_connected.py                                          40:40:A7:F2:C6:04 (witch is my phone's 
Mac address)

morrolinux, 2020-02-02:
Maybe I've got an idea. I changed the bluetooth event script, so:
1) move to mpradio folder and `git pull` to download the update
2) run this command `sudo cp install/usr/lib/udev/bluetooth /usr/lib/udev/bluetooth` 
3) reboot and test

GiannisDorokidis, 2020-02-02:
I did
"cd mpradio" then "git pull"
And then "sudo cp.... bluetooth"

morrolinux, 2020-02-02:
...and?

GiannisDorokidis, 2020-02-02:
the same bluetooth connected no playback from it

morrolinux, 2020-02-02:
as always I need the status of mpradio.service to see if anything's changed

GiannisDorokidis, 2020-02-02:
https://pastebin.com/ebVrLBSx

morrolinux, 2020-02-03:
I was able to reproduce the issue and devised a fix. you need to try it out:
run: `cd mpradio/install && git pull && git checkout embedding_bt_dev_lookup && sudo bash install.sh`
let me know

GiannisDorokidis, 2020-02-03:
It does nothing the status led is showing that its blinking randomly but it shuts it self down
I think I need to format the sd card because I cant get to the ssh shell at all

GiannisDorokidis, 2020-02-03:
and no music streamed while its doing that

morrolinux, 2020-02-03:
in min.py
https://github.com/morrolinux/mpradio-py/blob/db1d1502594dcbd414ff631e81373fbce0729dce/src/mpradio.py#L81
comment out this line with a `#` if that's the case

morrolinux, 2020-02-03:
you should have the time to `ssh pi@<your-raspi> 'sudo systemctl disable mpradio'` 
otherwise you can simply mount the sdcard in your pc and go straight editing the file mentioned above

GiannisDorokidis, 2020-02-03:
I put the sd card on my pc and I looked at overlays and didn't find it 
I also tried to use the search bar for files

morrolinux, 2020-02-03:
If you're on windows I think you have to enable/install something to get to read ext4 partitions.. otherwise you could use a linux virtual machine for example.
If you're fast enough, you could run the command above BEFORE mpradio starts and just after the network is present. you'd need a monitor connected to the Pi to help yourself with this.
Otherwise you can start over from scratch

GiannisDorokidis, 2020-02-03:
I re-installed it here is the output https://pastebin.com/wnhU1rpN

GiannisDorokidis, 2020-02-03:
https://pastebin.com/X3MGvWxs mpradio service

morrolinux, 2020-02-03:
You gotta be on the test branch I was talking about earlier...
`cd mpradio/install && git pull && git checkout embedding_bt_dev_lookup && sudo bash install.sh`
then test

GiannisDorokidis, 2020-02-03:
> cd mpradio/install && git pull && git checkout embedding_bt_dev_lookup && sudo bash install.sh

output https://pastebin.com/6fY74DJT

morrolinux, 2020-02-03:
loooks all right. does it work?

GiannisDorokidis, 2020-02-03:
nice it works thank you so much!

GiannisDorokidis, 2020-02-03:
I found that same time when I live it playing for a lot of time it does a 1second loop of the sog thats playing
And Now that I have tried the bluetooth for the first time it lags some times
 If I enable btBoost will this stop?

closed 2020-02-04 by morrolinux