morrolinux/mpradio-py #55: Newst Raspbian newst mpradio install after install it shutdown

kiam001, 2020-02-14:


kiam001, 2020-02-14:
after power again it goes on and shutdown

morrolinux, 2020-02-14:
That's a duplicate of #50 - please always check opened issues first.
Which Raspberry model are you using?
I have to re-think/re-work the GPIO button controls: as of now the input logic is "inverted" from model to model. 
Meanwhile, a quick fix for this would be to edit https://github.com/morrolinux/mpradio-py/blob/master/src/gpio_remote.py#L45 
and remove the "not" in `if not input_state:`.
Only problem is now that you installed it, it will run on boot and you might not have time to edit that file. So you could:
1) be fast enough to disable mpradio service before it shuts down: `ssh pi@<your-pi-ip> 'sudo systemctl disable mpradio'`
or
2) plug your Pi's SD card in a Linux PC and edit that file
or
3) reinstall raspbian, git clone this repo, edit the file and THEN install 
or 
4) wait for me to fix it and attempt a new install from scratch. But I cannot give you a timeline on that yet as I'm quite busy for at least another week

morrolinux, 2020-02-24:
@kiam001 **Which Raspberry model are you using?**

kiam001, 2020-03-04:
Raspberry Pi 3

morrolinux, 2020-03-08:
Just fixed and tested, pull my latest commit.

closed 2020-03-08 by morrolinux