morrolinux/mpradio-py #56: The mpradio.service fails at start

Angenios, 2020-05-15:
[image]

Honestly, I have no idea on how this works. Or, at least, it is not really user-friendly. I rebooted my pi with an USB plugged in as you say, but after that the service doesn't work anymore. I have connected a wire to GPIO 4 in order to stream via FM, though it simply doesn't work. I have no idea what to do, I don't even know how to disinstall this thing. I admit I'm really ignorant in this matter, so I ask you a bit of help.

morrolinux, 2020-05-15:
Hi
Can you paste here your pirateradio.conf file?

Angenios, 2020-05-15:
> 
> 
> Hi
> Can you paste here your pirateradio.conf file?

Hello, thanks for reaching out, here you are:

```
[PIRATERADIO]
frequency=88.5
btGain=1.7
storageGain=1
treble=-6
output=fm
btBoost=false
implementation=pi_fm_adv

[PLAYLIST]
persistentPlaylist=true
resumePlayback=true
shuffle=true 

[RDS]
updateInterval=3
charsJump=6
stationName=MPRadio
rdsPattern=$ARTIST_NAME - $SONG_NAME
```

morrolinux, 2020-05-15:
looks alright, where is it placed on the filesystem?

Angenios, 2020-05-15:
> 
> 
> looks alright, where is it placed on the filesystem?

/home/pi/mpradio/install/pirateradio

morrolinux, 2020-05-16:
That should be moved to /pirateradio/pirateradio.conf

Angenios, 2020-05-16:
> 
> 
> That should be moved to /pirateradio/pirateradio.conf

Oh my god, it is working! Thank you a lot! I love you!

closed 2020-05-16 by Angenios