morrolinux/mpradio-py #57: I cannot perform playback controls or system commands

Angenios, 2020-05-16:
[image]

It denies me the permission to perform such commands even if I use `sudo`.

morrolinux, 2020-05-16:
Yeah in GNU/Linux sudo doesn't survive the > redirection operator. You should either use `sudo -i` to become root or use tee: `echo pause | sudo tee /tmp/mpradio_bt `

Angenios, 2020-05-17:
> 
> 
> Yeah in GNU/Linux sudo doesn't survive the > redirection operator. You should either use `sudo -i` to become root or use tee: `echo pause | sudo tee /tmp/mpradio_bt `
It still doesn't reply to any command, and I've noticed another problem but I'll open a new issue about it in order to make order between problems.