morrolinux/mpradio-py #58: The radio stops playing after an uncertain amount of time

Angenios, 2020-05-17:
Yesterday I left home during the evening and when I got home the radio was still on, but it only streamed a white noise, "bzzzzz". I checked the service status and it was OK, still active, so I tried stopping the service and starting it again. It worked, and the radio started to stream my playlist again.
But it doesn't end here, because then I went to sleep and when I woke up this morning I noticed the radio was still streaming white noise only ("bzzz"), so I stopped the service and I started it again and the radio started to stream my playlist again.
It seems that the radio stops streaming my playlist after a lot of time, let's say 4-5 hours or more. Is it normal? I thought it should stream my playlist on loop forever.

morrolinux, 2020-05-17:
Yeah that's a known PiFmAdv issue, it has to do with how the signal is being modulated (if I recall correctly) check out the former for more info

Angenios, 2020-05-17:
> 
> 
> Yeah that's a known PiFmAdv issue, it has to do with how the signal is being modulated (if I recall correctly) check out the former for more info

It seems the issue on PiFMAdv is still open. I can't help but accept this and restart the service every 3-4 hours. Thanks for explanation, anyway.

closed 2021-04-12 by morrolinux