morrolinux/mpradio-py #59: The radio plays music at a low speed

Angenios, 2020-05-17:
OK, so I think I've ruined something here. In a nutshell, I couldn't pair my Android 10 device with my Raspberry Pi 3 (Raspbian Buster) so I tried following [this guide](https://learn.pi-supply.com/make/fix-raspberry-pi-3-bluetooth-issues/) while the radio was still on, I rebooted the Raspberry and then I ended up having the radio play at a low speed (I think it's something like 0.50x) my playlist, plus adding a little white noise on the background (a really annoying whistle).
I tried uninstalling pi-bluetooth and installing it again (this time without blueman) from the Raspberry Pi, I rebooted my pi and the radio was still playing at a low speed, then I tried reinstalling mpradio and the problem is still here, but this time when I try to check service status it gives me a different error.
This is the error:
[image]

morrolinux, 2020-05-17:
This could have something to do with sampling rate, but I can't know for sure and it's probably best to just wipe and reinstall fresh, as this kind of things affect system-wide configuration across multiple components and debugging the issue could be just more troublesome and timewastimg than restarting clean

Angenios, 2020-05-17:
> This could have something to do with sampling rate, but I can't know for sure and it's probably best to just wipe and reinstall fresh, as this kind of things affect system-wide configuration across [...]

How do you suggest to uninstall mpradio?

morrolinux, 2020-05-17:
I wanted to say, re-flash your sdcard with the latest Raspbian and start from scratch

PixelPerfect, 2020-07-03:
I've experienced this exact issue to on a clean install of raspbian with a pi zero w

morrolinux, 2020-07-03:
It could be needed to blacklist some modules, pay attention at the last few lines of the installer output

aquaMatt900, 2021-02-10:
Blacklisting solved this for me (I'm not the original poster)

closed 2021-04-12 by morrolinux