morrolinux/mpradio-py #60: Problem with mpradio not brocasting

GiannisDorokidis, 2020-06-20:
Hello good afternoon I am having an issue with mpradio not brocasting anymore at any frequency that i write on pirateradio.config on the usb stick that I am using.
Here are some logs; 1 install ;
https://pastebin.com/50gN2rGZ
2 status of mpradio after reboot;
https://pastebin.com/U2kZgCRv
3 usb pirateradio.config file on usb stick (the only thing changed is the frequency and btBoost set to true
https://pastebin.com/qi4cPaLj

GiannisDorokidis, 2020-06-24:
ok tried to install this in debian 10 and it didnt work. I think this has the same issue as before
Logs;
1.Install https://pastebin.com/ypCq9Rd6
2. status https://pastebin.com/tys7UXc6

Marrley, 2020-06-26:
rpi4?

Marrley, 2020-06-26:
try to broadcast lower then 95.3MHz

GiannisDorokidis, 2020-06-26:
> rpi4?

raspberry pi zero w

GiannisDorokidis, 2020-06-27:
I changed the frequensy to 89.2 and it now broadcasts only on 88.0 fm
https://pastebin.com/U9fGY4bC

Also does the  raspbian version matter ?

GiannisDorokidis, 2020-06-27:
Its not working i am broadcasting at 89.2mhz 
https://pastebin.com/Jx8w3dX7
The first time i forgot to plug the usb to the pi and it broadcasted on 88.0 by defult
All of this is on raspbian 9

I tried this on raspbian 10 and it broadcasted again on 88.0 but it was playing normaly from the usb stick

Also does the raspbian version matter?

GiannisDorokidis, 2020-06-27:
ok found an issue that might be the answer to my problem 
https://github.com/morrolinux/mpradio-py/issues/56
but I dont know how i can move the file

1100101111, 2020-07-17:
Hey there!
I have the same issue on RasPi Zero W.

All the services are active and running.
The `pirateradio.config` is in /pirateradio on fs root.

I can pair & connect via BT and when the Pi boots I can hear that the frequency set up in .config gets "jammed" i.e. no "regular" signal from any stations as my receiver is right next to the RasPi. But it stays quiet, no music, just some buzzing...

Is file ownership relevant here? I see that `/pirateradio/pirateradio.config` and the folder itself is owned by the user pi while the other files are owned by root.

GiannisDorokidis, 2020-07-17:
>Is file ownership relevant here? I see that /pirateradio/pirateradio.config and the folder itself is owned by the user pi while the other files are owned by root.
The pirateradio.config file has to be moved from the pi user to the flash storage

GiannisDorokidis, 2020-07-17:
Is the pirateradio.config file on /home/pi/mpradio/install/pirateradio
?

1100101111, 2020-07-17:
nope, it's in `[filesystem root]/pirateradio/...`

GiannisDorokidis, 2020-07-17:
> nope, it's in `[filesystem root]/pirateradio/...`

I think this is the flash storage this is the right place for the file

1100101111, 2020-07-17:
ok strange, moved some audio files to `/pirateradio/` and they are found and played after boot. Then I can connect via BT and start some music. It takes a moment but then audio switches to (I guess) the BT audio sink... 
It is very low volume compared to playing audio from RasPi memory, but that's another issue...

GiannisDorokidis, 2020-07-17:
>It is very low volume compared to playing audio from RasPi memory, but that's another issue...

I know I have that to

The pirateradio.config file and the music have to be on the flash drive not on the pi for this to work

>It takes a moment but then audio switches to (I guess) the BT audio sink...

After Sometime it switches to the bt and if you noticed any audio stutters Just turn off wifi

closed 2021-04-12 by morrolinux