morrolinux/mpradio-py #63: Android App doesn't seemingly work at all

Knucklesfan, 2021-04-12:
Hi there. I downloaded the Android app to go with my Raspberry Pi Zero W with mpradio-py installed on it. mpradio-py is working great, but the android app is basically useless. The currently playing feature doesn't work seemingly at all, and the pause and play controls don't work either. Clicking on the settings tab causes the app to soft crash and occasionally hard crash. Could you possibly make this app open source? I'm sure the app isn't at the top of your priorities but I've worked with android apps before and I could possibly help make the app better. Absolutely love the mpradio-py project. I'm on a Google Pixel 3 running Android 11 unrooted. My Pi Zero is running the latest version of Raspbian with all updates and just mpradio and a driver for my screen included. Thanks!

morrolinux, 2021-04-12:
Yeah sorry I forgot to update both the readme AND the download link, but the App has already been open-sourced a while ago and updated quite a bit on the user expirience and error handling side of things. You can find the public sources here: https://github.com/morrolinux/mpradio-remote
And I've just updated the readme with the latest APK.
I'd be happy to have your feedback on that and I'll be very happy to accept any contributions :)

Knucklesfan, 2021-04-13:
Ohh, I was reading the readme.md and thought that the app was still closed source. I'll check it out and post any issues I find in it's repo.

closed 2021-04-13 by Knucklesfan